pimoroni/fanshim-python #2: Add flag for lite versions

monglet, 2019-06-26:
Consider adding flag to setup.py or install.sh warning of dependencies needed for lite versions (setuptools + psutil?)

Gadgetoid, 2019-06-28:
I didn't realise lite versions ( Raspbian Lite? ) didn't include `setuptools`. I'd dropped the fallback import to `distutils` since it can cause problems with our packaging.

Attempted to rectify in 31b58d748a6ad23368a403c77581450b6ba562d5

haggistech, 2019-06-30:
would also need python-psutil and python3-psutil :)

nmstoker, 2019-06-30:
Also needs **python3-dev** for the installation to work on a fresh Buster lite install:
`sudo apt-get install python3-dev`

Without doing that, you get:

```
Best match: RPi.GPIO 0.6.5
Processing RPi.GPIO-0.6.5.tar.gz
Writing /tmp/easy_install-q2f014__/RPi.GPIO-0.6.5/setup.cfg
Running RPi.GPIO-0.6.5/setup.py -q bdist_egg --dist-dir /tmp/easy_install-q2f014__/RPi.GPIO-0.6.5/egg-dist-tmp-0cd30dse
source/py_gpio.c:23:10: fatal error: Python.h: No such file or directory
 #include "Python.h"
          ^~~~~~~~~~
compilation terminated.
error: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

Gadgetoid, 2019-07-02:
I wonder if I should add `python-rpi.gpio` to the apt dependencies. Building a version from source from pip is a little overkill, and those packages tend to evade being updated- for better or worse.

haggistech, 2019-07-03:
Would probably make sense as you say then it will be picked up in updates etc

closed 2020-03-16 by Gadgetoid