pimoroni/fanshim-python #3: Button Press not registered

mtowara, 2019-06-27:
As per https://forums.pimoroni.com/t/fan-shim-button-always-held/10784
the button does not seem to work with Buster (I'm using a OctoPi image ontop of Buster Lite).

Same with python2 and python3.

mtowara, 2019-06-27:
And the answer is: The button is configured wrong, or there has been a hardware revision.
fanshim = FanShim(pin_button=3) works

Gadgetoid, 2019-06-28:
If you're using a Pi 4, this is because RPi.GPIO currently does not support setting pull up/down states on pins, so the button doesn't have the correct pull resistor set.

The button is additionally connected to BCM3 only to bring the Pi out of a low power state, and this workaround would conflict with any i2c devices you might use. If you're okay with that- then using BCM3 is fine- since it's part of the i2c bus it actually has a hardware pull-up resistor already in place, so RPi.GPIO not working doesn't effect it.

mtowara, 2019-06-28:
Ok, then this issue will propably fix itself in a few days time once RPi.GPIO is fixed :)

scottsweb, 2019-07-05:
Would this explain why I am unable to power down the fan? Running the example service the LED is behaving as expected but the fan remains on at all times.

Gadgetoid, 2019-07-05:
It would indeed!

mtowara, 2019-07-05:
@scottsweb  fyi, the fan part always worked fine for me. (as did the led)

scottsweb, 2019-07-05:
Thanks for the updates. 

Interesting that your fan control worked @mtowara - I wonder what is different for me there. I am using the `pip3` package... although I tried just the standard `pip` package too without much luck. I also couldn't get it installed from this repo on buster lite, Pi 4.

scottsweb, 2019-07-09:
For those following this issue you can now install the beta of RPi.GPIO:

`pip3 install --upgrade --pre rpi.gpio`

Unfortuantely it doesn't solve my particular issue which seems to be a lack of connection being made by the push fit header and pin 12.

wildestpixel, 2019-07-16:
The beta of rpi.GPIO fixes this for me on 2 Pi 4’s

thallada, 2019-07-19:
I have the beta installed but the fan still never powers down.

scottsweb, 2019-07-19:
@thallada I had the same problem with two shims. I think it must have been an issue with the Pi header being slightly off shape or something. The solution was to add a tiny drop of solder on pin 12 ... you can test it with a jumper wire first to make a good connection between the shim and the Pi.

closed 2019-10-21 by mtowara