pimoroni/fanshim-python #5: Changing LED colour does not work

jcx200, 2019-07-02:
When I try to change the LED colour on the shim for my Pi 4 nothing happens.

Is this also related to the similar issue with the shim button due to RPi.GPIO?

haggistech, 2019-07-03:
If you look at the automatic.py it has an example of setting colour. I have used mine to set the colour based on temperature so colour change does work :)

Gadgetoid, 2019-07-03:
Pull-ups aren't required for LED colour changes, so they should work on the Pi 4.

Are you seeing *anything* on your LED? It's possible that you have UART enabled (`enable_uart=1` in your `/boot/config.txt`) but IIRC this shouldn't necessarily stop the LED from working.

Make sure your Fan SHIM is also making good contact with the GPIO pins, you might have to give it a little wiggle since the friction-fit header isn't perfect. Run `python led.py` in the examples folder while you do this, it may suddenly burst into life.

jcx200, 2019-07-06:
I was able to get the LED working in terms of lighting up but it just stays green and I can't change it at all even with using the different examples within the example file

Gadgetoid, 2019-07-09:
Have you installed the Fan SHIM service? That will effectively take over control of the LED.

You might want to stop it with `sudo systemctl stop pimoroni-fanshim.service` and see if you wrestle control back.

If this is the case, I can add a command-line argument to disable LED ( and maybe even button? ) control in the service so you can do what you like in user scripts instead.

jcx200, 2019-07-11:
Ok I think I have found the proper issue. You were right in saying that the SHIM isn't making proper contact. I think i need to find a wayto get the shim to sit properly because I can never get good contact on the pins. Any  idea of the best way to mount it?

closed 2019-07-17 by jcx200