pimoroni/fanshim-python #11: Make sensors_temperatures() work for cpu-thermal or cpu_thermal

fryfrog, 2019-07-11:
Would it be possible for the `psutil.sensors_temperatures()['cpu-thermal'][0].current` line to use cpu_thermal if appropriate? I'm still digging into why the Arch Linux aarch64 5.2 kernel causes it to be cpu_thermal. :/

Gadgetoid, 2019-07-12:
I usually avoid supporting anything but Raspbian since there aren't enough hours in the day, but this is a releatively minor change so I've added it in 4177f389247d2db3e3b0a69115bdab8c8e2829d8

closed 2019-07-12 by fryfrog

puresilk, 2020-08-01:
FYI (Google brought me here): I'm experiencing the same problem with picockpit-client for two different **Raspbian** Buster instances, one was upgraded from Stretch. I'm using the same build of my package, and in one instance there is an underscore, and in one instance a dash.