pimoroni/fanshim-python #12: Not an issue, more a feature request

DaveB1977, 2019-07-11:
Hi, I'm new to RPi and loving it, recently installed the shim to help with temps but was wondering if it would be possible to invert the LED colours, so that it's green when "cool" and red when the fan is active i.e too hot. 

Steep learning curve with this gizmo and I'm slowly getting to grips with it :)

Ta.

Trotter73, 2019-07-12:
Hi,

In /examples/automatic.py

Look for the following and try swapping the 0,255,0 / 255,0,0 values, I think they correspond to Red,Green,Blue

def update_led(state):
    if args.noled:
        return
    if state:
        fanshim.set_light(0, 255, 0)
    else:
        fanshim.set_light(255, 0, 0)

I cant see mine at the moment so cant test :)

DaveB1977, 2019-07-12:
Hi, spot on, worked a treat! Many thanks for your help. 

I'm brand new to Pi, and having a blast working out the little things, hoping to get it up and running with OMV once it's updated for the Pi4 but having fun messing around. 

Thanks again!

closed 2019-07-12 by DaveB1977