pimoroni/fanshim-python #8: Review Arch Linux AUR package for fanshim?

fryfrog, 2019-07-11:
Hi,

I don't *have* a fanshim yet, but I've ordered a handful to go w/ the handful of Pi 4s I've ordered. I use and <3 Arch Linux, so I went ahead and created a fanshim package... which required a python-fanshim package which required a python-plasmalights package.

It'd be great if you could give it a look to confirm correctness. The python `automatic.py` script (renamed to fanshim) seems to work, but of course I can't test it yet.

https://aur.archlinux.org/packages/fanshim

https://aur.archlinux.org/packages/python-fanshim

https://aur.archlinux.org/packages/python-plasmalights

Also, it seems like the library name `plasma` in the plasmalights package is likely to conflict with other things. For example, in searching for just "plasma", I thought it was using some machine learning library.

fryfrog, 2019-07-11:
So the only thing I think I need to figure out is why `psutil.cpu_freq()` doesn't work, it returns `None` and the `/sys` folder is missing all the things a raspbian system has:

```
0 ✓ fryfrog@raspbian-armv7l /sys/devices/system/cpu/cpufreq $ ls -alh policy0/
total 0
drwxr-xr-x 3 root root    0 Jul 11 20:47 .
drwxr-xr-x 4 root root    0 Jul 11 19:10 ..
drwxr-xr-x 2 root root    0 Jul 11 20:42 stats
-r--r--r-- 1 root root 4.0K Jul 11 20:42 affected_cpus
-r-------- 1 root root 4.0K Jul 11 20:42 cpuinfo_cur_freq
-r--r--r-- 1 root root 4.0K Jul 11 20:42 cpuinfo_max_freq
-r--r--r-- 1 root root 4.0K Jul 11 20:42 cpuinfo_min_freq
-r--r--r-- 1 root root 4.0K Jul 11 20:42 cpuinfo_transition_latency
-r--r--r-- 1 root root 4.0K Jul 11 20:42 related_cpus
-r--r--r-- 1 root root 4.0K Jul 11 20:42 scaling_available_frequencies
-r--r--r-- 1 root root 4.0K Jul 11 20:42 scaling_available_governors
-r--r--r-- 1 root root 4.0K Jul 11 19:40 scaling_cur_freq
-r--r--r-- 1 root root 4.0K Jul 11 20:42 scaling_driver
-rw-r--r-- 1 root root 4.0K Jul 11 19:10 scaling_governor
-rw-r--r-- 1 root root 4.0K Jul 11 19:40 scaling_max_freq
-rw-r--r-- 1 root root 4.0K Jul 11 19:40 scaling_min_freq
-rw-r--r-- 1 root root 4.0K Jul 11 20:42 scaling_setspeed
```

```
0 ✓ fryfrog@arch-aarch64 /sys/devices/system/cpu/cpufreq $ ls -alh
total 0
drwxr-xr-x  2 root root 0 Jul 11 17:54 .
drwxr-xr-x 12 root root 0 Jan  1  1970 ..
```

https://archlinuxarm.org/forum/viewtopic.php?f=65&t=13268

^ This thread implies it may just be a limitation of the Arch aarch64.

fryfrog, 2019-07-11:
Using `linux-aarch64-raspberrypi-bin` for kernel fixes `psutil.cpu_freq()` and also switches `cpu_thermal` to `cpu-thermal`.

:/

Gadgetoid, 2019-07-12:
Honestly I know *nothing* about Arch packaging- I've been a Debian-based system user for more than two decades with only some brief brushes with ... the dark side 😆 

Thanks for packaging it up, though!

The issue with the "plasma" module in the "plasmalights" package is one of those unfortunate cases where we name a product something, and a pip package already exists with that name. *Grumble* I don't imagine there will be much overlap between users of these two packages, however.

closed 2019-07-12 by fryfrog