pimoroni/fanshim-python #13: Problem with psutil - Doesn't detect SoC temperature sensor if another temperature sensor exists.

trejan, 2019-07-12:
This is actually a problem with psutil and not your code but it does cause problems with the Fan SHIM. automatic.py won't work in some specific configurations.

The way psutil works for temperature sensors is that it'll check for /sys/class/hwmon/hwmon*/temp*_* and /sys/class/hwmon/hwmon*/device/temp*_*. If no temperature sensor is found only then will it look for /sys/class/thermal/thermal_zone*. On a RPi, the SoC temperature sensor is on that second code path for /sys/class/thermal/thermal_zone*.

The problem I've got is that I fitted a Fan SHIM to a RPi which has a DS3231 RTC attached to it which has an internal temperature sensor. The Linux driver exposes that temperature sensor in /sys/class/hwmon and causes psutil to never check /sys/class/thermal/thermal_zone* for the RPi SoC temperature sensor. automatic.py continually prints out "Warning: Unable to get CPU temperature!" because it can't find a cpu_thermal.

My workaround was to bypass psutil for reading the SoC temperature and directly read /sys/class/thermal/thermal_zone0/temp in get_cpu_temp()

Gadgetoid, 2019-07-12:
D'oh!

I guess I could add this as a failover if the other two cases don't succeed, I don't see it doing any harm.

Sorry I've not had any kernel-driver temperature sensors hooked up before, so I hadn't a hope in catching this. Thanks for posting a solution!

wkelley70, 2019-11-12:
Thanks for the information @trejan.  I added an Adafruit  DS3231  and ran into this issue.  Adding your code and removing the original CPU temp code from automatic.py fixed the CPU temp detection.

closed 2020-03-16 by Gadgetoid