pimoroni/fanshim-python #17: Need help

daddacorn2019, 2019-07-16:
I'm new to all this like new new know nothing of anything and everything I've done is from reading on the internet. Anyways I've downloaded everything I need for the fanshim it works and now I'm in Python and I've figured out how to type these commands in to get everything working the way I want and my problem lies that I have no idea how to save these commands cause as soon as I close Python it goes back to stock setting. What do I need to do to get these things implemented on my rasberry pi 4?

Gadgetoid, 2019-07-17:
You need to save these commands in a `.py` file and run that.

See, for example, https://github.com/pimoroni/fanshim-python/blob/master/examples/toggle.py

You could use the Thonny Python IDE to do this, or `nano` on the command-line.

daddacorn2019, 2019-07-17:
I'm sorry to be a nuisance but can you tell me how to save it, and so if this updates the signal if I'm reading it correctly does that mean I can run the commands in thonny and they will stay that way when I close thonny or do I have to save every command into the py file? Cause the only command I want to stick it the constant changing of the led I want my fan to stay on constant.

daddacorn2019, 2019-07-18:
So I ran what I wanted into thonny saved the file but as soon as I turn off my pi the commands doesn't stay . Any idea what I'm doing wrong

daddacorn2019, 2019-07-18:
All I'm trying to do is run the example led file so the led is constantly changing but I can't get it to stay on afterbi close thonny

daddacorn2019, 2019-07-18:
I did all of it saved it to a file but again as soon as I close thonny it stops the script, all I'm trying to do is runt the led example file, is there anyway to make it to an automatic script in the background or however to implement the script that when I close thonny after running it it will stay and turn back on even if I turn the pie off?

daddacorn2019, 2019-07-18:
So I save the script to that file and add python /path/to/file.py?

daddacorn2019, 2019-07-19:
Where can I locate that file at please?

grayerbeard, 2019-08-17:
One way to run Python automatically that allows you to more easily see what its doing.  Better when you want to play.  Is to instalkl and use "tmux".  Then edit rc.local to start the python in a tmux session at start up. see [https://robinhenniges.com/howto-raspberry-start-command-on-startup-with-tmux/](url)
Then you use the tmux a -t [sessionname]  command to have a peak at your code.

daddacorn2019, 2019-08-21:
Thank you so much seriously I appreciate it alot

closed 2019-08-21 by daddacorn2019