pimoroni/fanshim-python #18: noob-question about  "Getting Started with Fan SHIM"

Dimpelmoser, 2019-07-18:
I'm running a raspiblitz v1.2 on a 3B + board and trying to run a "fan shim" on it.

I have followed the following description several times:
https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-fan-shim. 
However, the "fan shim" always runs only in continuous operation and can not be controlled via the small pressure button.
The command:
    cd examples
    sudo ./install-service.sh 65 5
also remains after the installation without effect.

Please, what am I doing wrong?

Thanks

Gadgetoid, 2019-07-19:
Try (in terminal) - `sudo systemctl status pimoroni-fanshim.service` - and see what output you get.

It might be worth running the `automatic.py` file directly, too, and seeing if it produces an error or functions as expected:

```
python3 automatic.py
```

Dimpelmoser, 2019-07-19:
@Gadgetoid 
... thanks for your answer!
I'll try your proposals and give a feedback asap

Dimpelmoser, 2019-07-19:
... after my many previous attempts, I wanted to try the proposals with a "fresh" SD card.
On occasion, my raspiblitz blockchain has been corrupted and now it will take several days for my raspberry to be operational again.
Unfortunately, asap has become very relative, but I will be back ... for sure!

RazorWing, 2019-07-19:
I am having the same issue as Dimpelmoser. Exact steps followed, all appeared to be working then performed a reboot and the fan just runs constantly.

`automatic.py` does work (green led comes on and the fan stops). Running `sudo ./install.sh` and `sudo ./install-service.sh 65 5` does not seem to resolve the issue.

`sudo systemctl status pimoroni-fanshim.service` reports the service as "Inactive" and running `sudo systemctl start pimoroni-fanshim.service` just sits there with no change to the fan (running `sudo systemctl status pimoroni-fanshim.service` in another terminal window reports the service again as "Inactive").

It does appear to be a service related issue as `automatic.py` does work, just not on boot and startup of the service.

RazorWing, 2019-07-19:
I did manage to tweak the service using an example online which does the trick (until maybe an alternative solution comes along):

```
[Unit]
Description=Fan Shim Service
After=network.target

[Service]
Type=oneshot
WorkingDirectory=/home/pi/fanshim-python/examples
ExecStart=/home/pi/fanshim-python/examples/automatic.py --on-threshold 65 --off$
RemainAfterExit = yes

[Install]
WantedBy=multi-user.target
```

As an observation, I did notice running `./install-service.sh 65 5` with additional parameters such as `--noled` does not append to the end of the `ExecStart` line.

Dimpelmoser, 2019-07-28:
@Gadgetoid 
... sorry for the long delay, but now I'm back again.
Here are my results of your suggestions:

---------------

-> "Try (in terminal) - sudo systemctl status pimoroni-fanshim.service":

admin@PotzBlitz01:~/fanshim-python/examples $ sudo systemctl status pimoroni-fanshim.service                                                               
â pimoroni-fanshim.service - Fan Shim Service
   Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-07-28 14:17:20 BST; 33min ago
  Process: 26884 ExecStart=/home/admin/fanshim-python/examples/automatic.py --on-threshold 65 --off-threshold 55 --delay 2 (code=exited, status=1/FAILURE)
 Main PID: 26884 (code=exited, status=1/FAILURE)

Jul 28 14:17:20 PotzBlitz01 systemd[1]: pimoroni-fanshim.service: Failed with result 'exit-code'.
Jul 28 14:17:20 PotzBlitz01 systemd[1]: pimoroni-fanshim.service: Service hold-off time over, scheduling restart.
Jul 28 14:17:20 PotzBlitz01 systemd[1]: Stopped Fan Shim Service.
Jul 28 14:17:20 PotzBlitz01 systemd[1]: pimoroni-fanshim.service: Start request repeated too quickly.
Jul 28 14:17:20 PotzBlitz01 systemd[1]: Failed to start Fan Shim Service.
Jul 28 14:17:20 PotzBlitz01 systemd[1]: pimoroni-fanshim.service: Unit entered failed state.
Jul 28 14:17:20 PotzBlitz01 systemd[1]: pimoroni-fanshim.service: Failed with result 'exit-code'.
~
~
lines 1-13/13 (END)

---------------

-> "It might be worth running the automatic.py":

admin@PotzBlitz01:~/fanshim-python/examples $ sudo python3 automatic.py                       
Traceback (most recent call last):
  File "automatic.py", line 112, in <module>
    t = get_cpu_temp()
  File "automatic.py", line 62, in get_cpu_temp
    t = psutil.sensors_temperatures()
AttributeError: module 'psutil' has no attribute 'sensors_temperatures'

---------------

I am looking forward to hear from you again.

Samreay, 2019-09-04:
Just a random chiming in here, but @Dimpelmoser, I would try updating psutil and running again.

Try updating with

`sudo pip3 install psutil --upgrade`

Gadgetoid, 2019-11-05:
Looks like the `sensors_temperatures()` issue is a duplicate of https://github.com/pimoroni/fanshim-python/issues/33

Please report there if you've had any success! Or re-open here if you're still having trouble.

closed 2019-11-05 by Gadgetoid