pimoroni/fanshim-python #21: 65 degrees is probably too hot for the on threshold

ali1234, 2019-07-22:
If someone takes a fresh Pi 4 and installs a fan shim and nothing else, it is possible it will never spin at all. Any older revision of the Pi definitely won't reach 65 degrees unless it is put under extreme load. As mentioned on #19, this can lead to users thinking there is a problem, when in fact the system just isn't hot enough for the fan to spin. I would therefore consider lowering the on threshold to reduce support requests. :)

Gadgetoid, 2019-07-23:
I do wonder if there would be just as many "my fan spins all the time and never stops" support requests.

Should the installer, perhaps, force the user to make a choice, since no default is necessarily going to be correct for all ambient temperatures? I'm not sure!

ali1234, 2019-07-23:
I suspect there may be a happy medium somewhere. That said, yes, the installer should ask, or at least make it clear what the expected mode of operation is.

jankais3r, 2019-07-25:
Totally disagree with the premise of this request. It's like if you bought a fire alarm and then complained that it's not sounding when there is no fire. The point of buying the fan is not to have another source of noise, the point is to prevent overheating the CPU - and 65 degrees is perfectly valid threshold for that. I would also argue that people who buy this product know better than to be confused why the fan does not spin up.

grayerbeard, 2019-08-17:
If you use PWm ... See https://github.com/pimoroni/fanshim-python/issues/22 then you can set the lower temperature to about 58 and the fan will come on at a slow speed which might reassure the worriers but even then on light load it takes quite a while to get that warm.

However I agree that strictly the fan settings could be even higher without any harm done (say) 68 and 72 as then most people will rarely see it spin and your still well clear of the throttling speed.

HeadyWook, 2020-05-18:
Are you all talking about degrees Fahrenheit? On the Pimoroni's [tutorial](https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-fan-shim) they say, "In our testing, an on threshold of 65°C, an off threshold of 55°C, and a delay of 2 work pretty well. Type the following in the terminal to run the installer for the automatic script:"
[image]
For example, 55 degrees Celsius is 131 degrees Fahrenheit and 65 degrees Celsius is 149 degrees Fahrenheit. 
I use Fahrenheit so I agree that the threshold is high. If the fan was on 24/7, what would the average low temperature be? If the fan was off, what would the average high temperature be? If we knew this we would also calculate the standard deviation for both low and high temp and set the parameters in terms of standard deviations from the low and high temp.
Maybe:
When temp reaches the average low plus two standard deviations, then stop fan.
When temp reaches the average high minus two standard deviations, then start fan.

Any feedback?

druck13, 2020-05-18:
There are a verity of ways of using the fanshim, the standard setup of 65 off and 55 on (and the new Pi4 firmware that reduces power usage), means that when idle the fan will be off and the Pi's temperature will probably be in 50's depending on the ambient temperature. A modest amount of load will cause it to reach 65C and trip the fan, which will go off when the Pi returns to idle.

Another way is to set the off threshold to a value much lower than the normal idle temperature with the fan on, so the fan stays on continuously. This keeps the Pi's temperature as low as possible at all times, but I've found the little fans have a limited lifetime, I've had two fail in under a year.

I find the standard set up runs too hot, and the fan on continuously is overkill as overnight it can go down to as low as 35C. So I'm using a third way of making the off threshold just above normal fan off idle temperature (50C for me) and the fan on about 5 degres above this. This means the fan comes on when there is any load, but stays off most of the time when idle, giving a good compromise between the life of the Pi, and the life of the fan.

The only issue with this is the LED goes from green to red over a very short range, and I prefer it to show green at idle with the fan off through to red at full load with the fan on. So I've forked my own version which has two extra parameters which control the green and red LED temperatures independently of the off and on thresholds (if set, otherwise defaults to the same as off and on).

I didn't stop there though, I noticed that sometimes over night (and also just after booting) the temperature is lower than my normal low value for green, and when doing benchmarks for several hours, can be higher than my normal temperature for red. So I added a bit of code to change the colour from blue to green in the below normal range, and from red to magenta in the above normal range. So most of the time its between green and red as expected, but if its outside of that you can see it as a glance - handy if the monitor is switched to a different computer.

Gadgetoid, 2020-08-14:
It's going to vary quite considerably by region and ambient room temperature, so I'm not sure there's any benefit to changing the default so much as attempting to better explain how to identify what number anyone should use.

I'm happy to drop down to 60C. In a cold room (my bme280 is reading ~15 degrees!) my Pi is already idling at 52C, so I feel any lower and there's a good chance the fan would just be on all the time.

closed 2023-01-20 by Gadgetoid