pimoroni/fanshim-python #24: No control on Fan/LED

feh123, 2019-07-25:
I followed the assembly instructions and installed the service. However vcgencmd measure_temp gives value 46C and pimoroni-fanshim.service is set at 50C on threshold (with no LED). 

Yet fan and LED still on. Where have I gone wrong? Thanks

TheGreatestJannet, 2019-07-25:
I am experiencing the same issue. I have set it to 65 5 and vcgencmd measure_temp shows 40 degrees but the fan is still spinning.

feh123, 2019-07-26:
Yes - good spot. On reading the thread I did not want to try the solder trick on pin 12 so wiggled the shim around on the pins and the fan stopped as it should - threshold 55 temp and from vcgencmd 52. Only issue now is the fan LED as its still green and not red. That does not seem to be right - or am I missing something? Thanks for your post.

feh123, 2019-07-27:
My idea of flexing the board to find a good position does not work any more. I am back to the situation that fan is either on or off. I am using buster on a RPI4. I have updated buster every day in the hope it may solve the issue but I think this is probably down to poor contacts with the pins. Is there a solution to this?

CaNsA, 2019-07-28:
I understand that this will be an unpopular suggestion, but based on my experience I found that bending pins 2,4,6,8,10,12 outwards just a tiny amount prior to fitting the shim resolved my spotty contact issue.

feh123, 2019-07-28:
I do not have the courage to bend the pins. I like the idea of fixing to the motherboard. However, while the fan was on (ie normally!) and at the closet point between the fan cowling and the the pins I pushed a suitably sized plastic spacer. Sized so that it was just large enough to stop the fan. This seems to have worked and the fan behaves (currently) according to the fanshim.service settings. No LED but a minor matter. If this flops I will try your motherboard standoff solution.

thul, 2019-09-13:
Is this still broken? i recently got a fan shim for my pie that's running LibreElec, im trying to talk to the pins directly via RPi.GPIO, but yea... not working

Gadgetoid, 2019-11-05:
`RPi.GPIO` should be long since fixed at this point. Closing this for now, but please feel free to re-open if you're still encountering problems.

closed 2019-11-05 by Gadgetoid