pimoroni/fanshim-python #25: Fan state only persists during FanShim object lifetime

tomhosking, 2019-07-26:
I've set up a quick cron job to check the CPU temp and update the fan/LED state accordingly - but once the script exists, the fan switches back on again (regardless of what state was set by the script). I can force the LED state to persist using `plasma.set_clear_on_exit(False)`, is there an equivalent for the fan?

My full script is below - the fan state will persist for as long as the `sleep()` at the end .

```
from subprocess import PIPE, Popen
from time import sleep

import plasma

def get_cpu_temperature():
    """get cpu temperature using vcgencmd"""
    process = Popen(['vcgencmd', 'measure_temp'], stdout=PIPE)
    output, _error = process.communicate()
    output = output.decode()
    val = output[output.index('=') + 1:output.rindex("'")]
    return float(val)

temp = get_cpu_temperature()

print('Temp is', temp)

from fanshim import FanShim
fanshim = FanShim()

plasma.set_clear_on_exit(False)

if temp > 60:
        print('Enabling fan')
        fanshim.set_fan(1)
        print(fanshim.get_fan())
else:
        print('Disabling fan')
        fanshim.set_fan(0)
        print(fanshim.get_fan())

light = 255 * max(0, min(1, (temp-40)/50))

fanshim.set_light(light, 0, 255-light)
sleep(30)
```

Gadgetoid, 2019-08-02:
There's nothing explicitly calling for the GPIO pins to be reset at exit, so I wonder if there's a bug or regression in `RPi.GPIO` that causes this to happen depsite `GPIO.cleanup()` never being called.

Your best bet is to write a script that doesn't exit, so it can "own" the GPIO pins and keep them in a consistent state. If you're intending to run your script periodically anyway then there aren't really any drawbacks to this approach.

tomhosking, 2019-08-05:
OK thanks - I'll go with an infinite loop instead!

closed 2019-11-29 by tomhosking