pimoroni/fanshim-python #27: Unable to use service with raspbian lite

Kyosfonica, 2019-07-27:
I am not sure if the library is working at all because the fan never stops but the service is definitively not working. It is hanging when you start it from command line and when you check the status after a reboot this is what I get:
```pimoroni-fanshim.service - Fan Shim Service
   Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
```
I noticed that the installer installs both python2 and 3. I do not know why it need 2 versions of python to work.

Any guidance is more than welcome.

Gadgetoid, 2019-08-01:
I think I just ran into exactly this problem- the code was structured in such a way that- if psutil was already installed- it would never install the Fan SHIM library. I've fix it now, but for your setup you should just be able to `sudo pip3 install fanshim` to get it up and running. Assuming you're having the problem I just discovered!

gh3rann, 2019-08-12:
Hello, did you manage to solve this? I am having the very same problem.

gh3rann, 2019-08-13:
> I think I just ran into exactly this problem- the code was structured in such a way that- if psutil was already installed- it would never install the Fan SHIM library. I've fix it now, but for your [...]

I am gettting: 
pi@raspberrypi:~/fanshim-python $ sudo pip3 install fanshim
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: fanshim in /usr/local/lib/python3.7/dist-packages/fanshim-0.0.2-py3.7.egg (0.0.2)
Requirement already satisfied: plasmalights in /usr/local/lib/python3.7/dist-packages/plasmalights-0.0.1-py3.7.egg (from fanshim) (0.0.1)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from plasmalights->fanshim) (0.7.0)

And still I cannot seem to fin out the issue.

Gadgetoid, 2019-11-05:
@gh3rann did you find a solution to your problem? It might be that something else is causing the `automatic.py` to fail, and you should probably run the example file directly to see if it outputs any errors.

gh3rann, 2019-11-05:
Hello Philip,
I did not try any longer, since I could not figure it out after many
attempts and days. I bought a different passive-massive cooling method
which is working according to my needs. Good luck to anyone with this issue.

Gadgetoid, 2019-11-06:
@gh3rann thanks for letting me know- I'm sorry it didn't work out for you!

closed 2019-11-06 by Gadgetoid

BButner, 2021-02-24:
Just as a heads up, I ran into the exact same issue. I think it just randomly happened after I ran an apt upgrade.

The way I was able to get it back up and running was to modify the service file entry, and comment out the "After" line:

`sudo nano /etc/systemd/system/pimoroni-fanshim.service`

Then here's my new entry:
```
[Unit]
Description=Fan Shim Service
#After=multi-user.target

[Service]
Type=simple
WorkingDirectory=/home/pi/FanShim/examples
ExecStart=/usr/bin/python3 /home/pi/FanShim/examples/automatic.py --on-threshold 65 --off-threshold 55 --low-temp 55 --high-temp 65 --delay 2 --brightness 255 
Restart=on-failure

[Install]
WantedBy=multi-user.target
```

Once you change the file:

`sudo systemctl daemon-reload`
`sudo systemctl restart pimoroni-fanshim.service`

And that got the service back up and running. (So now I can actually set it up properly now that I know how to get it functioning!)