pimoroni/fanshim-python #31: Change of behaviour

Trotter73, 2019-08-01:
Hi,

I have been running the new RPi bootloader for a week or so now.

[Linky](https://www.raspberrypi.org/forums/viewtopic.php?f=117&p=1502216#p1502216)

The big change is that 3V3 and 5V pins remain powered on following sudo halt, of course the upshot of this is that the fan will continue to run after you have shutdown.

Would be possible to have a trigger in the service that will turn the fan off when the system is shutdown cleanly, or maybe a cron job ? 

Me.

Gadgetoid, 2019-08-01:
The fan is default-on, so unless the Pi is capable of asserting a pin in the halt state then it's not possible.

You could try adding the following to `/boot/config.txt`:

```
dtoverlay=gpio-poweroff,gpiopin=18
```

Trotter73, 2019-08-02:
Hi, so if the fan is in the off state and the pi is shut down it will it will start again ?

I wasn't hoping to change state in halt, I was thinking that during the shutdown process we could set the fan to stop. 

What does the option do, set the state of the pin when you shutdown, like pressing the button in the shim ? Just curious...

I'll try this when I have eye on the pi.

Thanks for the reply !

Gadgetoid, 2019-11-06:
Looking back over the documentation for `gpio-poweroff` I don't think it can help us in this case. Fan SHIM requires the control pin to be pulled low in order to turn off the fan and `gpio-poweroff` is active-high, with some serious caveats for active-low use. You can run `dtoverlay -h gpio-poweroff` to see the docs.

Short of trying to include some power board to cut power to the Pi (which will definitely turn the fan off) when it is shut down, I don't think there's an easy way to keep a pin asserted low after the system has shut down.

closed 2019-11-22 by Trotter73