pimoroni/fanshim-python #32: code review/help: c++ code for fanshim

daviehh, 2019-08-01:
Following up on https://github.com/pimoroni/fanshim-python/issues/19, I made a quick C++ code for the fanshim (no LED/button yet). Anyone interested in review the code for bugs, etc? Note that it's very experimental/rough around the edges,  it may make the fan run all the time, or using excessive cpu/memory/disk, which may or may not cause hardware damage, so the code is only for testing/playing now, and I wouldn't run it unattended and I do not think it's ready for serious use for now. Thanks a lot!

github repo: https://github.com/daviehh/fanshim-cpp

Gadgetoid, 2019-08-19:
It's not the most useful of feedback and there doesn't even seem to be a lot of useful guidance out there for it, but for basic GPIO access you should drop the BCM2835 library and switch to libgpio- the new standardised way of accessing GPIO on Linux.

Things like libbcm2835 are holdouts from a bygone era where register bitbanging was the only reasonable way to do GPIO on the Pi and are only really useful for performance-sensitive GPIO access, and some of the Pi specific features that don't have a good cross-platform alternative yet.

There's no guarantee the BCM2835 library will be maintained in the future and its contemporary - wiringPi - was recently deprecated.

Making this switch will start to be on people's TODO lists, so depending on what you want to learn/do with C++ on the Pi in the future it might be worth a look!

daviehh, 2019-08-19:
@Gadgetoid Thanks a lot, that's very helpful! I did look into libgpio before but at that time it did not seem to be very well documented...

Gadgetoid, 2019-11-06:
@daviehh yeah- I found the same with libgpio. Documentation was scarce, as were real-world examples of it being used. The more code out there using it, the more working examples people will have to reference, and hopefully that will get what's been a very slow ball rolling with respect to its adoption.

Good effort switching over to it- I've added a link to your library in our README.

closed 2023-01-20 by Gadgetoid