pimoroni/fanshim-python #38: "ImportError: cannot import name FanShim" on Buster

m0ahn, 2019-08-10:
I am getting the following error on Raspbian 10 Buster:

> Traceback (most recent call last):
>   File "fanshim.py", line 1, in <module>
>     from fanshim import FanShim
>   File "/home/pi/fanshim.py", line 1, in <module>
>     from fanshim import FanShim
> ImportError: cannot import name FanShim

The code is:

> from fanshim import FanShim
> fanshim = FanShim()
> fanshim.toggle_fan()

Run via:

> python fanshim.py

The install happened without errors??

druck13, 2019-08-11:
On 10/08/2019 21:03, m0ahn wrote:
> I am getting the following error on Raspbian 10 Buster:
> 
>     Traceback (most recent call last):
>     File "fanshim.py", line 1, in
>     from fanshim import FanShim
>     File "/home/pi/fanshim.py", line 1, in
>     from fanshim import FanShim
>     ImportError: cannot import name FanShim

Enter python (run python or python3) and do a help("modules") and look 
for fanshim. If it isn't there try running the install script again.

Cheers
---David
-- 
Email: druck@druck.org.uk
Phone: +44-(0)7974 108301

m0ahn, 2019-08-11:
Yes it is there David. As expected - the install script ran without errors (and i've already run it again)

m0ahn, 2019-08-13:
For info my sys.path is showing:

```
python -c "import sys; print(sys.path)"
['', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-arm-linux-gnueabihf', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/usr/local/lib/python2.7/dist-packages', '/usr/local/lib/python2.7/dist-packages/fanshim-0.0.2-py2.7.egg', '/usr/lib/python2.7/dist-packages']
```

and

```
python3 -c "import sys; print(sys.path)"
['', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/usr/local/lib/python3.7/dist-packages', '/usr/local/lib/python3.7/dist-packages/fanshim-0.0.2-py3.7.egg', '/usr/lib/python3/dist-packages']
```

Dimpelmoser, 2019-08-15:
> I am getting the following error on Raspbian 10 Buster:
>
> ```
> Traceback (most recent call last):
>   File "fanshim.py", line 1, in <module>
>     from fanshim import FanShim
>   File "/home/pi/fanshim.py", [...]

I do know nothing, but you can try the following:

-->    **_sudo_** python fanshim.py

...  maybe it helps ... maybe !? ..

grayerbeard, 2019-08-16:
Is there an issue with FanShim. V fanshim without capital letters?

m0ahn, 2019-08-18:
No joy with either:
# FanShim V fanshim
# sudo python fanshim.py

grayerbeard, 2019-08-19:
Try instead ti run the toggle.py example.

Gadgetoid, 2019-08-19:
If you're running a python file entitled `fanshim.py` then `import fanshim` will import your `fanshim.py` rather than the `fanshim` module. This is one of Python's most ridiculous footguns since one would normally assume that there's no way you'd ever reasonably want to recursively import a file into itself.

You should rename your code from `fanshim.py` to literally anything else and it will work.

m0ahn, 2019-08-19:
Amazing - you got it. No wonder this was proving so difficult to pinpoint.

Thankyou @Gadgetoid

closed 2019-08-19 by m0ahn