pimoroni/fanshim-python #39: Install not completed, stuck@ "systemctl restart --no-pager pimoroni-fanshim.service"

gh3rann, 2019-08-12:
Hello, I am trying to install the service by "./install-service.sh --on-threshold 60 --off-threshold 50 --delay 2". Yet the script seems to not go through and the screen is black. I tried with "sudo bash -x ./install-service.sh --on-threshold 60 --off-threshold 50 --delay 2" to debug and this is what happens:

```
pi@raspberrypi:~/fanshim-python/examples $ sudo bash -x ./install-service.sh --on-threshold 60 --off-threshold 50 --delay 2
+ ON_THRESHOLD=65
+ OFF_THRESHOLD=55
+ HYSTERESIS=5
+ DELAY=2
+ PREEMPT=no
+ POSITIONAL_ARGS=()
+ NOLED=no
+ NOBUTTON=no
+ BRIGHTNESS=255
+ OLD_THRESHOLD=
+ OLD_HYSTERESIS=
+ SERVICE_PATH=/etc/systemd/system/pimoroni-fanshim.service
+ '[' -f /usr/bin/python3 ']'
+ '[' -f /usr/bin/pip3 ']'
+ [[ 6 -gt 0 ]]
+ K=--on-threshold
+ case $K in
+ ON_THRESHOLD=60
+ shift
+ shift
+ [[ 4 -gt 0 ]]
+ K=--off-threshold
+ case $K in
+ OFF_THRESHOLD=50
+ shift
+ shift
+ [[ 2 -gt 0 ]]
+ K=--delay
+ case $K in
+ DELAY=2
+ shift
+ shift
+ [[ 0 -gt 0 ]]
+ set --
+ EXTRA_ARGS=
+ '[' '' == error ']'
+ '[' '' == error ']'
+ '[' no == yes ']'
+ '[' no == yes ']'
+ '[' no == yes ']'
+ '[' '' == '' ']'
+ '[' '' == '' ']'
+ cat
Setting up with:
Off Threshold:  50 C
On Threshold:   60 C
Delay:          2 seconds
Preempt:        no
Disable LED:    no
Disable Button: no
Brightness:     255

To change these options, run:
sudo ./install-service.sh --off-threshold <n> --on-threshold <n> --delay <n> --brightness <n> (--preempt) (--noled) (--nobutton)

Or edit: /etc/systemd/system/pimoroni-fanshim.service

+ read -r -d '' UNIT_FILE
++ pwd
++ pwd
+ printf 'Checking for rpi.gpio>=0.7.0 (for Pi 4 support)\n'
Checking for rpi.gpio>=0.7.0 (for Pi 4 support)
+ python3 -
+ '[' 0 -ne 0 ']'
+ printf 'rpi.gpio >= 0.7.0 already installed\n'
rpi.gpio >= 0.7.0 already installed
+ printf 'Checking for Fan SHIM\n'
Checking for Fan SHIM
+ python3 -
+ '[' 1 -ne 0 ']'
+ printf 'Installing Fan SHIM\n'
Installing Fan SHIM
+ pip3 install fanshim
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: fanshim in /usr/local/lib/python3.7/dist-packages/fanshim-0.0.2-py3.7.egg (0.0.2)
Requirement already satisfied: plasmalights in /usr/local/lib/python3.7/dist-packages/plasmalights-0.0.1-py3.7.egg (from fanshim) (0.0.1)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from plasmalights->fanshim) (0.7.0)
+ printf 'Checking for psutil\n'
Checking for psutil
+ python3 -
+ '[' 0 -ne 0 ']'
+ printf 'psutil already installed\n'
psutil already installed
+ printf '\nInstalling service to: /etc/systemd/system/pimoroni-fanshim.service\n'

Installing service to: /etc/systemd/system/pimoroni-fanshim.service
+ echo '[Unit]
Description=Fan Shim Service
After=multi-user.target

[Service]
Type=simple
WorkingDirectory=/home/pi/fanshim-python/examples
ExecStart=/home/pi/fanshim-python/examples/automatic.py --on-threshold 60 --off-threshold 50 --delay 2 --brightness 255
Restart=on-failure

[Install]
WantedBy=multi-user.target'
+ systemctl daemon-reload
+ systemctl enable --no-pager pimoroni-fanshim.service
+ systemctl restart --no-pager pimoroni-fanshim.service

```

Please notice I tried the "sudo usermod -a -G gpio pi", but nothing happens. 
Can you please point me in the right direction?

gh3rann, 2019-08-12:
If I run it without the debug part this is the output:
``
pi@raspberrypi:~/fanshim-python/examples $ sudo ./install-service.sh --on-threshold 60 --off-threshold 50 --delay 2
Setting up with:
Off Threshold:  50 C
On Threshold:   60 C
Delay:          2 seconds
Preempt:        no
Disable LED:    no
Disable Button: no
Brightness:     255

To change these options, run:
sudo ./install-service.sh --off-threshold <n> --on-threshold <n> --delay <n> --brightness <n> (--preempt) (--noled) (--nobutton)

Or edit: /etc/systemd/system/pimoroni-fanshim.service

Checking for rpi.gpio>=0.7.0 (for Pi 4 support)
rpi.gpio >= 0.7.0 already installed
Checking for Fan SHIM
Installing Fan SHIM
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: fanshim in /usr/local/lib/python3.7/dist-packages/fanshim-0.0.2-py3.7.egg (0.0.2)
Requirement already satisfied: plasmalights in /usr/local/lib/python3.7/dist-packages/plasmalights-0.0.1-py3.7.egg (from fanshim) (0.0.1)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from plasmalights->fanshim) (0.7.0)
Checking for psutil
psutil already installed

Installing service to: /etc/systemd/system/pimoroni-fanshim.service

``

druck13, 2019-08-12:
On 12/08/2019 19:25, gh3rann wrote:
> Hello, I am trying to install the service by "./install-service.sh 
> --on-threshold 60 --off-threshold 50 --delay 2". Yet the script seems to 
> not go through and the screen is black.

Nothing will appear on the screen after installing, all you should see 
is the LED on fan turn on, and be somewhere between green and red 
depending on the temperature. To check if the service is running, do a

systemctl status pimoroni-fanshim.service

---druck

-- 
Email: druck@druck.org.uk
Phone: +44-(0)7974 108301

gh3rann, 2019-08-12:
Hello Druck, indeed I tried this and:

`pi@raspberrypi:~ $ sudo systemctl status pimoroni-fanshim.service
● pimoroni-fanshim.service - Fan Shim Service
   Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
`

I was trying to status/stop/start/ctrl+c/status:
`pi@raspberrypi:~ $ sudo systemctl stop pimoroni-fanshim.service
pi@raspberrypi:~ $ sudo systemctl status pimoroni-fanshim.service
● pimoroni-fanshim.service - Fan Shim Service
   Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
pi@raspberrypi:~ $ sudo systemctl start pimoroni-fanshim.service
^C
pi@raspberrypi:~ $ sudo systemctl status pimoroni-fanshim.service
● pimoroni-fanshim.service - Fan Shim Service
   Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
pi@raspberrypi:~ $
`

And it keeps being inactive. Also there is no led lit on the fan itself.

gh3rann, 2019-08-13:
Also
_pi@raspberrypi:~/fanshim-python/examples $ python automatic.py
Traceback (most recent call last):
  File "automatic.py", line 2, in <module>
    from fanshim import FanShim
  File "/home/pi/fanshim-python/examples/fanshim.py", line 1, in <module>
    from fanshim import FanShim
ImportError: cannot import name FanShim_

Gadgetoid, 2019-08-19:
The latter is to be expected- since Fan Shim runs with (and is installed for) Python 3- you'd have to test with `python3 automatic.py` and see if you get any useful output.

gh3rann, 2019-08-19:
Hello & thank you for your response. The output its preetty much the same:
pi@raspberrypi:~/fanshim-python/examples $ alias python37
alias python37='/usr/bin/python3.7'
pi@raspberrypi:~/fanshim-python/examples $ python37 -m pdb automatic.py
> /home/pi/fanshim-python/examples/automatic.py(2)<module>()
-> from fanshim import FanShim
(Pdb)

Gadgetoid, 2019-08-19:
I'm guessing in your case the library just hasn't been installed for your Python version- I merged a fix earlier today which might help if you "git pull" the repository and run "examples/install-service.sh" again.

gh3rann, 2019-08-19:
Thank you for your suggestion. Same thing - please see attached log (http://s000.tinyupload.com/index.php?file_id=03746890479925919913)

I am including the part where I am checking that the library was installed before/after refreshing the repo.

Gadgetoid, 2019-08-19:
Do you still see "inactive (dead)" when you `sudo systemctl status pimoroni-fanshim.service`?

I'm starting to wonder if the code is fine, but the physical connection between Fan SHIM and your Pi is flaky. It might be worth trying to adjust the fit of Fan SHIM, giving it a wiggle to make sure the friction-fit header is making good contact.

gh3rann, 2019-08-19:
Hello again. Unfortunately, yes:

> pi@raspberrypi:~ $ sudo systemctl status pimoroni-fanshim.service
> ● pimoroni-fanshim.service - Fan Shim Service
>    Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
>    Active: inactive (dead)
> pi@raspberrypi:~ $ date
> Mon 19 Aug 2019 03:04:45 PM CEST

Actually its a bit weird since at the moment the fan is all the time on. I would assume its really fitting the GPIO.

Gadgetoid, 2019-08-20:
Possible thought- is the exec bit set on `automatic.py`? Since the unit file for the service is running `ExecStart=/home/pi/fanshim-python/examples/automatic.py` directly.

The fan should be on-by-default which means it's getting power- the enable pin is pulled up. It not turning off could be indicative of a flaky connection.

It might be worth trying to toggle it more directly by running something like this in python3:

```python
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.OUT)
GPIO.output(18, 0) # SHould turn fan off
GPIO.output(18, 1) # Should turn fan on
```

This should help narrow down if it's the software or the hardware/connection causing the problem.

gh3rann, 2019-08-20:
Thank you for your reply. It took a while since I couldnt phisically check on the fan until now. 

I did what you suggested (just adding the  GPIO.cleanup() at the end), but it did not shutdown the fan at all. It just kept going. I kind of saved the output to the file below. 

http://s000.tinyupload.com/index.php?file_id=03743350135855647087

Gadgetoid, 2019-11-05:
The fan defaults to "on", so it should only turn off when pin 18 is actively driven low. My example above probably needed to look more like:

```python
import time
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.OUT)
GPIO.output(18, 0)  # Should turn fan off
time.sleep(5)       # Wait 5 seconds
GPIO.output(18, 1)  # Should turn fan on
time.sleep(5)       # Wait 5 seconds
```

So that you actually had time to see the fan stop, rather than the pin just toggling really fast and doing ... nothing much.

closed 2020-03-16 by Gadgetoid