pimoroni/fanshim-python #40: Fan doesn't stop at all!

rossirm, 2019-08-19:
Hi, there, I have bad luck with my FanShim cause it doesn't stop though the off-threshold and keeps spinning all the time. Have you any ideas how I can solve this issue?
Note: I am new to programming in Python!

Gadgetoid, 2019-08-20:
Might be worth disabling the service and seeing if you can control the fan directly with Python:

```
sudo systemctl stop pimoroni-fanshim.service
```

Then run Python:

```
sudo python
```

And type:

```python
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.OUT)
GPIO.output(18, 0) # Should turn fan off
GPIO.output(18, 1) # Should turn fan on
```

Alternatively you can save the latter into a `test.py`, comment out the "on" portion and run it.

rossirm, 2019-08-20:
There is no effect of this solution but can I ask you where do you know the pin number of the fan which is pin 12 (BCM18)? Can you give me some link, if you have seen it somewhere, please!?

Gadgetoid, 2019-08-21:
You can find pin assignments documented on my website here - https://pinout.xyz/pinout/fan_shim

This not working suggests you might have an intermittent/poor connection between your Fan SHIM and Pi. It might be worth double-checking it's mounted securely since the friction-fit header isn't perfect.

The best way to try this is by running something like:

```
import signal
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.OUT)
GPIO.output(18, 0)
signal.pause()
```

And then adjusting the Fan SHIM gently until the GPIO connection is made and the fan stops.

- Also note that if you run the code in my above resposne verbatim it will very quickly turn the fan off/on and you will see no effect.

rossirm, 2019-08-23:
Thank you so much!
My problem is solved after I attached on top of GPIO a T-Cobbler Breakout Kit for Raspberry Pi and realized that the real problem in my case comes from lack of good contact between Fanshim and RPi 4.
It works properly now! :)

closed 2019-08-23 by rossirm