pimoroni/fanshim-python #42: Fan Light won't turn on.

TheGreatestJannet, 2019-08-21:
Hi I am using a Pi 4 and I have installed the ./install.sh and then ran sudo ./install-service.sh --on-threshold 65 --off-threshold 55 --delay 2. This was worked correctly and the fan turns on and off at the thresholds however the light on the fan will not turn on. Anyone know why or how to solve this? I have installed it from the GitHub repo and not the stable package.

Gadgetoid, 2019-08-22:
Might be worth giving the shim a wiggle to make sure it's making good electrical contact.

TheGreatestJannet, 2019-08-22:
Thank you that seems to be the issue. It seems like it needs to be very precise for the light to even turn on and just jogging it can make it turn off again.

closed 2019-08-22 by TheGreatestJannet