pimoroni/fanshim-python #44: Using this controller with different hardware

RoryDH, 2019-08-31:
Hey! I happen to already have a with a black and red wire which connect to ground and 3.3/5v. 

Does the fan shim hardware have any special electronics that allow this script to control it?
Or are there other pins I can use to achieve the same effect with the fan I have.

Thanks

Gadgetoid, 2019-09-20:
Since the Pi's GPIO pins can't source or sink the kind of current needed to run a fan, you would need a transistor, relay or other such circuitry to convert the Pi's "logic level" digital outputs into something capable of switching the fan on/off.

RoryDH, 2019-09-22:
Ah! I understand.
Would be nice if the schematic for fanshim was available however I know you guys likely want people to purchase...

closed 2019-09-22 by RoryDH