pimoroni/fanshim-python #48: Falcon PiCap

donwooton, 2019-10-09:
My PiCap uses GPIO 18/19 to operate.  I have a conflict with fanshim on GPIO 18 (pin 12).  Is there a workaround?  I am happy having the fanshim run constantly or manually controlled.

Gadgetoid, 2019-10-14:
What happens to the fan/PiCap when attempting to use them together without any Fan SHIM software running?

Fan SHIM has an onboard pull-up to 3v3 on this pin, which could certainly interfere.

If you're up for it, you could cut the trace leading from the fan control pin, but this may be a little extreme. If you're at all confident at soldering, it would always be possible to replace this with a wire from that pad across to the pull-up resistor- both fairly good targets for some bodge soldering.

You'd have to cut carefully across this trace on the top side of the board:
[image]

You could potentially also jump that over BCM4 which it's conveniently close to, which would change the layout to the following:
[image]

donwooton, 2019-10-14:
That’s great info. What wire should I use?

Gadgetoid, 2019-10-14:
My go-to for this sort of bodge is copper-enamel because it's insulated, and extremely thin/flexible. But any thin, stranded core wire will do (most solid core hookup wire is pretty thick and rigid to the point that you risk breaking a trace/pad off the board if it's knocked).