pimoroni/fanshim-python #49: LED always on after restart with --noled

SuNNjek, 2019-10-11:
I have the automatic example running as a service that automatically starts with my Pi. Since I don't want the LED to light up, I passed the `--noled` option to deactivate it.

Now after every reboot the LED is burning bright white which is quite annoying. Restarting the service fixes the problem temporarily, but it always turns on again after the next reboot.

Gadgetoid, 2019-10-14:
This is a tricky one, since `--noled` really means "don't touch the LED pins I might be using them for something else" and turning the LED off from whatever default state boot noise might have set it involves actively driving the LED.

We might need a `--led-mode` in addition to `--noled` which can be used to say "I want to drive the LED, but instead of making it colourful please turn it off."

SuNNjek, 2019-10-15:
I had a look through the last commits and it looks like this was already implemented in 058ebc3?

Gadgetoid, 2019-10-17:
That looks to be the case. Albeit it does cloud the meaning of “noled” a little, maybe I’ll wait and see if anyone complains!

closed 2020-03-16 by Gadgetoid