pimoroni/fanshim-python #50: Automatic.py stops working after few cycles

BassXX, 2019-10-24:
Hey there, I am using the automatic.py script from the examples.

In the first few cycles everything works as expected. However, for some reason it stops working later.

The fan won´t turn on at the treshold and the LED stays red. I checked the service and it says it is running.

After I reboot the pi, everything is back to normal untill it again stops triggering the fan after some cycles.

I am running Raspbian Buster on a RPi 4.

Any suggestions on what I could do? Thanks in advance. Regards

grayerbeard, 2019-10-25:
I would test it with this little bit of code (attached and below)  which also allows you to test use of PWM.
You may find a pwm rate that nicely keeps your Pi 4 cool enough.
That way you know the hardware is OK.
(I have made this by copy and paste from a more complicated pro, apologies if there is an error. ref: https://www.smalle.uk/r-pi-4-blog)

Then you can move on to figure what the cause is.
[pwm_test.zip](https://github.com/pimoroni/fanshim-python/files/3772151/pwm_test.zip)

GPIO.setup(18, GPIO.OUT)
pwm_out = GPIO.PWM(18,1)
pwm_out.start(0)
##############  change following two lines 60 means 60%, 6hz usually works OK
pwm_speed  = 60
pwm_freq = 6
################
pwm_out.ChangeFrequency(self.pwm_freq)
pwm_out.ChangeDutyCycle(self.pwm_speed)

Gadgetoid, 2020-08-14:
Closing for now since it's a stale issue- but did you ever get to the bottom of this @BassXX? I'm afraid I'm stumped- but would probably run `automatic.py` manually for a while to see if it throws errors.

closed 2020-08-14 by Gadgetoid