pimoroni/fanshim-python #53: Could the `automatic.py` just be part of the pypi package?

fryfrog, 2019-11-05:
It'd be great if the `automatic.py` could just be part of the `fanshim` pypi package, part of setup.py as a setutools entry point? Maybe even just call it `fanshim`? :)

Gadgetoid, 2019-11-06:
It’s possible, but it’s a little bit harder to maintain and update (necessitates a new Pypi release with every tweak) and since you need the bash installer to set it up as a service anyway I’m not sure users would gain anything.

There’s certainly an argument for versioning it- though- at the moment it’s changed quite substantially with no easy route to a recent “stable” release.

fryfrog, 2019-11-06:
Fair enough, and it is easy enough to pin a version to a hash so no worries. Thanks for thinking about it and explaining. :)

closed 2019-11-06 by fryfrog