pimoroni/fanshim-python #54: Create new release?

fryfrog, 2019-11-05:
The 0.0.2 release is from late June and there have been a lot of commits since then, could you do a new release? :)

Gadgetoid, 2019-11-06:
I use releases to track changes to the 
Python library itself- creating a release with every version uploaded to Pypi. In this case there has actually been a small change to the library code, so it’s probably due an update!

Gadgetoid, 2019-11-06:
Added release v0.0.2 which I'd missed when deploying that version of the Pypi package- https://github.com/pimoroni/fanshim-python/releases/tag/v0.0.2

Quite a lot has changed since then - https://github.com/pimoroni/fanshim-python/compare/v0.0.2...master

But the only unreleased change to the library is this: https://github.com/pimoroni/fanshim-python/commit/0246d230c5aa4f0ec6bafefedcc8ecf1cbaff9d3#diff-e919354f6bfffc05fc44eb1a4eacb906 which replaces https://github.com/pimoroni/fanshim-python/pull/6

Gadgetoid, 2019-11-06:
v0.0.3 push'd!

closed 2019-11-06 by Gadgetoid

fryfrog, 2019-11-06:
Thanks a bunch! :)