pimoroni/fanshim-python #55: get_fan() returns 0, fan is on.

AlfredDaw, 2019-11-06:
I'm not a python (or really any code) kind of person at all, but I made this script to check simply if the fan is on:

```py3
#!/usr/bin/env python3
from fanshim import FanShim
fanshim = FanShim()

status = 3

status = fanshim.get_fan()

print('status is: {}'.format(status))

if status == 1:
        print("The fan is on.")
else:
        print("The fan is off.")
```

It says "The fan is off" every time I run it, but the fan is on and has been on since I installed it yesterday.
I've installed the daemon, which is running, and tried using automatic.py too.

When running this directly, it doesn't turn off, either.
```py3
#!/usr/bin/env python3
from fanshim import FanShim
fanshim = FanShim()

fanshim.set_fan(False)

print("The fan is off, now.")
```

Have I done something wrong, or is this a different issue, maybe?

Gadgetoid, 2019-11-06:
This is an interesting sort-of-bug- basically when you run `get_fan()` you're reading the pin state of the pin that drives the fan. If you haven't already set a specific pin state then it's possible that the pull-resistor that keeps the fan "on" is not sufficient enough to read as a "high" when you read "get_fan()".

Alternatively it could be indicative of a flaky connection between the Fan SHIM and your Pi.

Also `get_fan()` should probably return `True/False`.

mooncresta, 2019-11-07:
I have a similar issue , I can control the LED, but not the fan now. Used to work fine (in service mode and direct from python) and not changed anything (I think) apart from the rpi-eeprom update a week ago, Since then the fan is always on regardless of whether the temp is high and even forcing via set_fan doesn't work. Checked Pin fit and seems fine + LED control is fine.
Is it possible something has changed in the eeprom update, I can't see anything about BCM18 the fan control pin ? https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

grayerbeard, 2019-11-07:
Seems to me that fanshim is a rather complicated way to control a GPIO pin when you are trying to fault find hardware.  Far better to just try controlling it direct with GPIO. Or what I suggested in https://github.com/pimoroni/fanshim-python/issues/50

mooncresta, 2019-11-07:
I tried that - updated the comment and then deleted it as found the issue (it was the connection and soldering the shim on fixed it so nothing to do with eeprom updates)
Was testing with gpio as well e.g.
gpio -g write 18 (1 or 0) 
works fine - (as PI seems to have the wiringPi gpio utility installed by default)

closed 2019-11-08 by Gadgetoid