pimoroni/fanshim-python #56: Fan turns back on if `set_light()` is part of the script

thunderBiscuit, 2019-11-16:
Take the following two scripts for example:

```python
# script A: simply shutting down the fan
from fanshim import FanShim
fanshim = FanShim()
fanshim.set_fan(False)
```

```python
# script B: shutting down the fan and setting a colour
from fanshim import FanShim
fanshim = FanShim()
fanshim.set_fan(False)
fanshim.set_light(255, 0, 0)
```

The first one will shut down the fan properly, while the second one will shut it down for a split second (I assume) and turn it right back on when the script exits.

Any idea how I could gracefully change the colour of the LED while also turning the fan off/on?

Gadgetoid, 2019-12-11:
Are you sure that the first example actually works? In both cases the fan should turn back on again because the Python script will release control of the GPIO upon exit and the pin should return to being an input.

There's no simple way around this, since the GPIO shouldn't be owned or asserted by a script that's not running. One approach is to ensure your Python script continues to run, and use it like a service to control the fan/light. We're lacking a good example of this in action and I think we should have one since this is a common question.

thunderBiscuit, 2019-12-12:
Weird. Indeed the first example works (keeps the fan off after the script returns) for me! I tested this very script this morning:

```python
from fanshim import FanShim

fanshim = FanShim()

fanshim.set_fan(False)
print('The fan is now off')
```

And the fan does stay off after the script prints to console and returns. @Gadgetoid would you like me to open a separate issue for this?

Gadgetoid, 2019-12-14:
I'll have to try this myself- I can't fathom any reason why the scripts would differ. Either way, keeping a program running is the "technically correct" way to assert the fan as off, but since on the Raspberry Pi it's normal to completely bypass any proper userspace GPIO interface and write the registers directly ... well things don't really work how they should.

thunderBiscuit, 2019-12-18:
All right well thanks for looking into it and clearing things out! It does look like it's a small inconsistency probably coming from proper use of userspace GPIO then. Let me know if there is anything I could do to help test further!

closed 2020-03-16 by Gadgetoid