pimoroni/fanshim-python #58: Fanshim doesn't seem to install correctly on my RasPi4 running Ubuntu

spiff72, 2019-12-08:
Maybe this was a stretch, but I was hoping to get this working on my RasPi4 running the Ubuntu 19.10 image (64 bit).  For now it just runs the fan all the time, but I would love to have the automatic.py functionality working under Ubuntu as well as Raspbian.

Thanks!

grayerbeard, 2019-12-08:
Does the Python GPIO work under Ubunto or can it be added, I would think a short Python script tryying ti use GPIO would establish if its there.?
Putting the aporopriate words inti Google led me here which suggests that is where to start   https://www.raspberrypi.org/forums/viewtopic.php?t=110092

spiff72, 2019-12-09:
I have installed it, and installed the RPi.GPIO package, but when I attempt to follow the example on the main page, I get an error:

ubuntu@ubuntu:~/fanshim-python$ python3
Python 3.7.5 (default, Nov 20 2019, 09:21:52) 
[GCC 9.2.1 20191008] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from fanshim import FanShim
>>> fanshim = FanShim()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/fanshim-0.0.3-py3.7.egg/fanshim/__init__.py", line 31, in __init__
RuntimeError: Not running on a RPi!
>>>

spiff72, 2019-12-09:
I have done some digging, and it seem that this error can sometimes occur if a user is not a member of the necessary group - in this case, I tried adding the ubuntu user to the gpio group, rebooted, and still no joy.  If i run python3 initially with sudo, then the command doesn't generate an error, but that seems like the nuclear option.  Are there other groups that my user may need to be part of for this to work correctly?

spiff72, 2019-12-09:
OK - I tried installing the fanshim service, which installs successfully as far as I can tell, but it never turns the fan on and the LED stays off.  I suspect that it is unable to read the temperature of the cpu, and therefore cannot control the fan properly.

rei-vilo, 2019-12-09:
I went through the same problem, albeit on a Raspberry Pi 3 with Debian 9.11.

+ Installed `psutil` didn't provide `sensors_temperatures()`

Check with Python

```
import psutil
psutil.sensors_temperatures()
```

In case of an error, install and upgrade `psutil` to the latest version.

```
pip install --upgrade psutil
pip3 install --upgrade psutil
```

And check again with Python

```
import psutil
psutil.__version__
'5.6.7'
psutil.sensors_temperatures()
{'cpu-thermal': [shwtemp(label='', current=55.148, high=None, critical=None)]}
```

+ But still not working

When I run `python automatic.py`, the fan runs fine. But it doesn't work when installed as a service.

```
systemctl | grep pimoroni
● pimoroni-fanshim.service                                                                                                loaded failed failed    Fan Shim Service        
```

Gadgetoid, 2019-12-09:
The tinkering here is indicative of why we only support Raspbian. Keeping everything working on one (or three, if you account for a couple of previous Raspbian versions) OS is time consuming enough without tackling any more. There simply aren't enough hours in the day.

That said, I'm always happy to draw attention to forks, ports or other projects that support our boards in other languages and on other platforms. And, time provided, to lend some of my experience.

spiff72, 2019-12-09:
I understand - I was just trying to find out if others had run into this same issue with Ubuntu.

FYI - I had another very odd thing that happened while trying to get those Ubuntu installations to boot for the first time.  It seems that if the Fanshim is pressed to tightly against the pi's gpio header, u-boot was reading strange characters as keypresses and the boot process was being interrupted.  The 3D printed case I had the Pi installed (specifically designed to support the fanshim) was squeezing the shim against the header and causing this.

Gadgetoid, 2019-12-11:
This is strange- are you saying it was causing characters to be produced via the UART pins/serial terminal? If those pins are left floating but enabled then I could see Fan SHIM introducing noise

spiff72, 2019-12-11:
That does seem to be what is happening.  bent the pins slightly on the header last night to try to improve the connection from pins to the fanshim, and got a fairly tight connection.  Oddly this is what seems to cause issues most of the time at bootup.  When I did this, I was getting the interruption in the boot process.  

I then removed the fanshim, rebooted, and added a line to the /boot/firmware/syscfg.txt file (enable_uart=0) based on suggestions from Dave Jones (waveform80) the Ubuntu team via the raspberrypi.org forums.  I then reinstalled the fanshim, rebooted, and it booted correctly.  I even tried again while applying downward pressure on the fanshim during boot to see if I could get the interrupted boot again, and couldn't get it to fail.  

This is the thread I was discussing the issue in:
https://www.raspberrypi.org/forums/viewtopic.php?f=131&t=258532

With the above adjustments, and a change to automatic.py to replace the get_cpu_temp() function as shown below, I was able to get the service to run correctly and operate the fan based on CPU temp.

`def get_cpu_temp():
    with open('/sys/class/thermal/thermal_zone0/temp', 'r') as f:
        return int(f.read()) / 1000`

The above is needed because psutil isn't reading the temps from the right location under ubuntu:
`ubuntu@ubuntu:~/fanshim-python/examples$ sudo python3
Python 3.7.5 (default, Nov 20 2019, 09:21:52) 
[GCC 9.2.1 20191008] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil
>>> psutil.__version__
'5.5.1'
>>> psutil.sensors_temperatures()
/usr/lib/python3/dist-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon1/temp1_input'
  RuntimeWarning)
/usr/lib/python3/dist-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon1/temp2_input'
  RuntimeWarning)
/usr/lib/python3/dist-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon2/temp1_input'
  RuntimeWarning)
/usr/lib/python3/dist-packages/psutil/_pslinux.py:1223: RuntimeWarning: ignoring FileNotFoundError(2, 'No such file or directory') for file '/sys/class/hwmon/hwmon2/temp2_input'
  RuntimeWarning)
{}
>>> `

I previously tried forcing psutil to update to 5.6.7, but this didn't help either.

Gadgetoid, 2020-02-25:
I wonder if it's worth distilling your findings down to some options in the `install-service.sh` script. It looks like we could potentially give the user a choice between `psutil` and just reading the raw temperature value from a given path ( `/sys/class/thermal/thermal_zone0/temp` also works on Raspbian).

This, combined with `/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq` for CPU Frequency would give a psutil-less alternative for instances where psutil doesn't work.

spiff72, 2020-02-25:
Sorry - I have abandoned the FanShim in favor of some passive cooling options.  This was mainly because the FanShim had gotten very noisy (weird vibrations and buzzing).

druck13, 2020-02-25:
On 25/02/2020 14:36, spiff72 wrote:
> Sorry - I have abandoned the FanShim in favor of some passive cooling 
> options. This was mainly because the FanShim had gotten very noisy 
> (weird vibrations and buzzing).

The bearings in the tiny fan can fail, it starts getting nosier and may 
eventually stop altogether. Pimoroni sent me a replacement fan, free of 
charge.

Cheers
---David
-- 
Email: druck@druck.org.uk
Phone: +44-(0)7974 108301

Gadgetoid, 2020-03-16:
Closing in favour of https://github.com/pimoroni/fanshim-python/issues/73

closed 2020-03-16 by Gadgetoid