pimoroni/fanshim-python #60: Fanshim service issue Raspbian 9.11

rei-vilo, 2019-12-09:
I went through a problem similar to #58, albeit on a Raspberry Pi 3 with Raspbian 9.11.

When I run `python3 automatic.py`, the fan runs fine. 

But it doesn't work when installed as a service.

```
$ sudo ./install-service.sh --on-threshold 65 --off-threshold 55 --delay 2
Setting up with:
Off Threshold:  55 C
On Threshold:   65 C
Delay:          2 seconds
Preempt:        no
Disable LED:    no
Disable Button: no
Brightness:     255

To change these options, run:
sudo ./install-service.sh --off-threshold <n> --on-threshold <n> --delay <n> --brightness <n> (--preempt) (--noled) (--nobutton)

Or edit: /etc/systemd/system/pimoroni-fanshim.service

Checking for rpi.gpio>=0.7.0 (for Pi 4 support)
rpi.gpio >= 0.7.0 already installed
Checking for Fan SHIM
Fan SHIM already installed
Checking for psutil
psutil already installed

Installing service to: /etc/systemd/system/pimoroni-fanshim.service
● pimoroni-fanshim.service - Fan Shim Service
   Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-12-09 18:46:34 CET; 409ms ago
 Main PID: 6041 (python3)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/pimoroni-fanshim.service
           └─6041 python3 /home/pi/fanshim-python/examples/automatic.py --on-threshold 65 --off-threshold 55 --delay 2…55

Dec 09 18:46:34 myPiTwo.local systemd[1]: Started Fan Shim Service.
```

Checking with `systemctl`

```
$ systemctl | grep pimoroni
● pimoroni-fanshim.service   loaded failed failed    Fan Shim Service        
```                   

I went through different solutions, but the pimoroni-fanshim.service is still not working.

Thank you for your help!                                           

## Already checked: `psutil` —from #33

+ Installed `psutil` didn't provide `sensors_temperatures()`

Check with Python 3

```
import psutil
psutil.sensors_temperatures()
```

In case of an error, install and upgrade `psutil` to the latest version 

```
$ pip3 install --upgrade psutil
```

And check again with Python

```
import psutil
psutil.__version__
'5.6.7'
psutil.sensors_temperatures()
{'cpu-thermal': [shwtemp(label='', current=55.148, high=None, critical=None)]}
```

## Already checked: GPIO —from #39

Run on Python 3

```
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.OUT)
GPIO.output(18, 0) # Should turn fan off
GPIO.output(18, 1) # Should turn fan on
```

Gadgetoid, 2019-12-09:
Does `systemctl status pimoroni-fanshim` give any useful output- Python errors, or other messages in the log?

rei-vilo, 2019-12-09:
Here is the result of `systemctl status pimoroni-fanshim`:

> $ systemctl status pimoroni-fanshim
> ● pimoroni-fanshim.service - Fan Shim Service
>    Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Mon 2019-12-09 19:18:41 CET; 30min ago
>  Main PID: 10716 (code=exited, status=1/FAILURE)
> 
> Dec 09 19:18:41 myPiTwo.local systemd[1]: pimoroni-fanshim.service: Unit entered failed state.
> Dec 09 19:18:41 myPiTwo.local systemd[1]: pimoroni-fanshim.service: Failed with result 'exit-code'.
> Dec 09 19:18:41 myPiTwo.local systemd[1]: pimoroni-fanshim.service: Service hold-off time over, scheduling restart.
> Dec 09 19:18:41 myPiTwo.local systemd[1]: Stopped Fan Shim Service.
> Dec 09 19:18:41 myPiTwo.local systemd[1]: pimoroni-fanshim.service: Start request repeated too quickly.
> Dec 09 19:18:41 myPiTwo.local systemd[1]: Failed to start Fan Shim Service.
> Dec 09 19:18:41 myPiTwo.local systemd[1]: pimoroni-fanshim.service: Unit entered failed state.
> Dec 09 19:18:41 myPiTwo.local systemd[1]: pimoroni-fanshim.service: Failed with result 'exit-code'.

rei-vilo, 2019-12-10:
The only workaround I've found so far is to launch

```
python3 fanshim-python/examples/automatic.py &
```

Not very clean, especially if the Raspberry Pi reboots.

rei-vilo, 2019-12-18:
Any answer?

Gadgetoid, 2020-02-25:
I had no idea what the problem was in this case, so I was at a loss to providing a fix. However it *might* be related to- https://github.com/pimoroni/fanshim-python/issues/66

I've since added a fix for the executable path, so re-running `install-service.sh` might help.

rei-vilo, 2020-02-25:
The fix from #66 didn't solve the issue. 

```
$ systemctl | grep pimoroni
● pimoroni-fanshim.service                                                                                                loaded failed     failed          Fan Shim Service             
```

DKeppi, 2020-06-12:
Same problem here, so i reinstalled it and now it works :)

Gadgetoid, 2020-08-14:
@rei-vilo did you ever get to the bottom of this?

rei-vilo, 2020-08-14:
A clean install (OS + Fanshim utility) solved the issue on the Raspberry Pi 4B.

closed 2020-08-14 by rei-vilo

FlorentLM, 2021-02-23:
Having the exact same issue here.

I am not conviced a reinstall of the OS should be an acceptable solution to this though