pimoroni/fanshim-python #64: Looking for help to changing the LED Color from Green to Blue

prawnz, 2020-01-02:
I am using the command sudo ./install-service.sh --on-threshold 65 --off-threshold 55 --delay 2 
which works however I do want to change the LED to turn from blue to red instead to better reflect from cool to hot. Anyone have any ideas how do I do that easily?

Gadgetoid, 2020-01-02:
The file `automatic.py` is what gets installed as the service. You can modify it to customise its behaviour.

LED colour is currently calculated as a portion of the hue (from 0 to 120 degrees) in the hue-saturation-value colour space:

[image]

Here's the code that does it:

```
def update_led_temperature(temp):
    led_busy.acquire()
    temp = float(temp)
    temp -= args.off_threshold
    temp /= float(args.on_threshold - args.off_threshold)
    temp = max(0, min(1, temp))
    temp = 1.0 - temp
    temp *= 120.0
    temp /= 360.0
    r, g, b = [int(c * 255.0) for c in colorsys.hsv_to_rgb(temp, 1.0, args.brightness / 255.0)]
    fanshim.set_light(r, g, b)
    led_busy.release()
```

I believe the range you want is probably 360 degrees to 240 and, right off the top of my head, you should be able to get that by changing the coloursys line to:

```
    r, g, b = [int(c * 255.0) for c in colorsys.hsv_to_rgb(1.0 - temp, 1.0, args.brightness / 255.0)]
```

If Red-Purple-Blue isn't your cup of tea, you could expand the range of the colour sweep so that it goes Red-Yellow-Green-Blue. In that case you ignore the above line, and instead change the scale of the sweep around the colour space turning:

```
    temp *= 120.0
```

To:

```
    temp *= 240.0
```

This should move counter-clockwise around from 0 degrees (red) to 240 degrees (blue).

prawnz, 2020-01-04:
ah good.. instructions. I got it working now. Thanks

closed 2020-01-04 by prawnz