pimoroni/fanshim-python #67: Request: HASSIO add-on for Fan-SHIM

KC-inDomus, 2020-02-16:
As you know, Raspberry Pi is commonly used for home automation; Home Assistant is one of the mostly adopted ecosystems on Rapsberry. 
When installed on Raspbian, Fan SHIM could be implemented with no headache; btw when you install HASSIO distribution, actually there'a no Pimoroni custom component or add-on for that. An implementation like this will be greatly appreciated by thousands of HASSIO users that what to us Fan-SHIM on their Raspberry Pi with HASSIO distribution on it.

Gadgetoid, 2020-02-24:
I don't disgree- but alas I'm stretched too thin already to figure out the ins-and-outs of HASSIO and implement/support an add-on for it. Nevertheless, I would appreciate and endorse anyone who makes it happen.

haggistech, 2020-03-16:
you can add the following to your configuration.yaml file to give you CPU temp and a Fan Switch

# Raspberry Pi Cooling Fan
sensor:
  - platform: command_line
    name: CPU Temperature
    command: "cat /sys/class/thermal/thermal_zone0/temp"
    unit_of_measurement: "°C"
    value_template: '{{ value | multiply(0.001) | round(1) }}'
    scan_interval: 10
switch:
  - platform: rpi_gpio
    ports:
      18: RPI Cooling Fan
climate:
  - platform: generic_thermostat
    name: RPI Cooling Fan Controller
    heater: switch.rpi_cooling_fan
    target_sensor: sensor.cpu_temperature
    min_temp: 40
    max_temp: 80
    ac_mode: true
    target_temp: 40
    cold_tolerance: 0.1
    hot_tolerance: 0.1
    min_cycle_duration:
      seconds: 120
    keep_alive:
      minutes: 5
    initial_hvac_mode: "cool"