pimoroni/fanshim-python #69: exit service

matteos1, 2020-02-18:
sudo systemctl status pimoroni-fanshim.service
● pimoroni-fanshim.service - Fan Shim Service
   Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-02-18 21:19:55 CET; 9ms ago
 Main PID: 2341 (python3)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/pimoroni-fanshim.service
           └─2341 /usr/bin/python3 /root/fanshim-python/examples/automatic.py --on-threshold 40 --off-threshold 30 --delay 2 --brightness 255

Feb 18 21:19:55 DietPi systemd[1]: Started Fan Shim Service.
root@DietPi:~# sudo systemctl status pimoroni-fanshim.service
● pimoroni-fanshim.service - Fan Shim Service
   Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-02-18 21:19:57 CET; 511ms ago
  Process: 2347 ExecStart=/usr/bin/python3 /root/fanshim-python/examples/automatic.py --on-threshold 40 --off-threshold 30 --delay 2 --brightness 255 (code=exit
ed, status=1/FAILURE)
 Main PID: 2347 (code=exited, status=1/FAILURE)

Gadgetoid, 2020-02-23:
Not sure what you're hinting at here, but if you want to know why the service might be failing then it's easier to diagnose when running it manually, ie:

```
/usr/bin/python3 /root/fanshim-python/examples/automatic.py --on-threshold 40 --off-threshold 30 --delay 2 --brightness 255
```

matteos1, 2020-02-24:
This is the error
root@DietPi:~# /usr/bin/python3 /root/fanshim-python/examples/automatic.py --on-threshold 40 --off-threshold 30 --delay 2 --brightness 255
Traceback (most recent call last):
  File "/root/fanshim-python/examples/automatic.py", line 103, in <module>
    t = get_cpu_temp()
  File "/root/fanshim-python/examples/automatic.py", line 50, in get_cpu_temp
    t = psutil.sensors_temperatures()
AttributeError: module 'psutil' has no attribute 'sensors_temperatures'

Gadgetoid, 2020-02-24:
You might possibly be able to `sudo apt remove python3-psutil` and then use pip to install the latest version. The version provided with Raspbian (buster) is 5.5.1, though, so maybe an `apt update && apt upgrade` would help.

matteos1, 2020-02-24:
pip3 install --ignore-installed psutil
now work

Gadgetoid, 2020-02-25:
Excellent! I should probably add a check for a compatible version of `psutil` to `install-service.sh`. Right now it only checks if it exists.

closed 2020-02-25 by Gadgetoid