pimoroni/fanshim-python #70: fanshim, omxplayer, and hyperpixel

dvschultz, 2020-02-21:
On my Pi 4B in CLI mode, I noticed when running the fanshim service videos played via omxplayer are filtered.

I think this might have to do with the green background used in the CLI when the service is running? can that be turned off?

without the service running:
[image]

with the service running:
[image]

Gadgetoid, 2020-02-22:
That's not a filter or an intentional green background. HyperPixel uses *all the pins* and by running Fan SHIM alongside it you're interfering with some of the colour data pins and basically inadvetently telling the screen to display different colours.

The LED control on Fan SHIM uses pins BCM 14 and BCM 15 which on HyperPixel correspond to bits 4 and 5 of the green colour channel. You could disable the Fan SHIM LED (which you probably can't see under the HyperPixel anyway) to try and mitigate this, but bear in mind that Fan SHIM's button maps to green 7 so wierd things may happen if you press it.

When you run `./install` to set up the service, you should pass the `--noled` and `--nobutton` args and it should, hopefully, fix this. 

The actual Fan control pin still shares the same GPIO as HyperPixel's LCD Chip Select, but I don't think that's a huge issue.

hacwa, 2020-02-29:
Unfortunately passing the --noled and --nobutton doesn't fix this.

Have also edited /etc/systemd/system/pimoroni-fanshim.service with the flags no change.
If you disable the fanshim service and then reboot the pi resets the colour to how it should be on the screen.

Running the following command below changes the colour on the screen immediately 
/home/pi/fanshim-python/examples/manual.py echo "off" > /tmp/fanshim

Restarting hyperpixel4-init.service seems to momentarily stop the fan running full speed, presume it's the shared GPIO pin.

Gadgetoid, 2020-03-16:
Just looked over the code and I can see that- indeed- the `--noled` and `--nobutton` options wont actually prevent `automatic.py` (or, more specifically, the FanSHIM library) from setting up those pins and trampling all over HyperPixel's required pin states.

It looks like the `FanShim` class will need `enable_led` and `enable_button` arguments that `automatic.py` can use, so that it can forego this setup.

closed 2022-02-14 by Gadgetoid