pimoroni/fanshim-python #72: FanShim on Raspberry Pi 4

rei-vilo, 2020-03-02:
I tried FanShim on a Raspberry Pi 4 and it doesn't work as expected:

+ Despite launching the Python service with default parameters, _ie._ `--off-threshold 55 --on-threshold 66`, the fan is always on, even when temperature = 28 °C

+ LED doesn't work, even when playing with the `python3 fanshim-python/examples/led.py example.

I went through #3, but how to configure the FanShim correctly for the Raspberry Pi 4?

The [product page](https://shop.pimoroni.com/products/fan-shim) advertises about the Raspberry Pi 4 but the [Getting Started with Fan SHIM](https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-fan-shim) doesn't offer any specifics. 

Thank you for your help.

rei-vilo, 2020-03-02:
+ **uname**
```
$ uname -a
Linux myPiFour 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
```

+ **RPi.GPIO**
```
>>> import RPi.GPIO as GPIO
>>> GPIO.VERSION
'0.7.0'
```

Gadgetoid, 2020-03-03:
I'd be inclined to believe this is a contact/electrical issue, do you get fan response/LED output if you run these examples while gently tilting the Fan SHIM in an effort to get better GPIO contact?

rei-vilo, 2020-03-03:
Thank you for your suggestion. Indeed, gently titling the FanSHIM solves the issue, alas as long as I use my finder. How to get a permanent solution without requiring a finger?

EGMSSE, 2020-03-05:
Yes, I am having the same issue. The shim itself is advertised as being able to make contact via "friction fitting". However, I have to gently apply pressure to it in just the right way to get the fan to react as expected, i.e., turn on and off, get the led to respond appropriately.

A solution that does not require soldering would be ideal. But I can confirm the friction fitting is not working for me with an RPi4

Gadgetoid, 2020-03-16:
Sometimes the Pi's GPIO can be bent by adding/removing HATs and correcting this can help.

Additionally, the manufacturing tolerance of the friction-fit header isn't perfect; we've shipped a lot of Fan SHIM units and issues are few and far between, but they happen. We're usually happy to try swapping your Fan SHIM out for another one if all else fails.

closed 2020-03-16 by Gadgetoid

jameselks, 2020-04-10:
I had the same problem - bending the pins out very slightly fixed it. Thanks!

EGMSSE, 2020-04-10:
yeah...VERY slightly, right?!

I actually used a plant lens that I had sitting around to better see how much each was bent. I'd imagine if you had one of those fancy jeweler's lenses that can be worn on the face and used hands free would be the best.

Glad I could help

dpd73, 2020-04-22:
I too found the LED wouldn't work unless I fiddled with the pins (used a fingernail!) then the LED appeared. Can be a bit temperamental. The button and fan worked fine. Guess that's the drawback of being friction fit.