pimoroni/fanshim-python #77: "Not running on a Pi" when running on a

voltagex, 2020-04-18:


hifi25nl, 2020-06-11:
I have a similar error running on archlinuxarm:

[audiolinux@raspberry ~]$ fanshim
Traceback (most recent call last):
  File "/usr/bin/fanshim", line 87, in <module>
    fanshim = FanShim()
  File "/usr/lib/python3.8/site-packages/fanshim/__init__.py", line 34, in __init__
    GPIO.setup(self._pin_fancontrol, GPIO.OUT)
RuntimeError: Not running on a RPi!

Gadgetoid, 2020-06-11:
This is an error produced by the `RPi.GPIO` library, you should attempt to update it to see if it fixes the problem:

```
python -m pip install rpi.gpio --upgrade
```

This may possibly also be related and suggests permissions issues: https://sourceforge.net/p/raspberry-gpio-python/tickets/171/

hifi25nl, 2020-06-11:
Ok, python-pip was not installed. Strange... 0.3 was working efore.

hifi25nl, 2020-06-23:
Today the problem is back... python-pip updated

  File "/usr/bin/fanshim", line 87, in <module>
    fanshim = FanShim()
  File "/usr/lib/python3.8/site-packages/fanshim/__init__.py", line 35, in __init__
    GPIO.setup(self._pin_button, GPIO.IN, pull_up_down=GPIO.PUD_UP)
RuntimeError: Not running on a RPi!

Gadgetoid, 2020-07-12:
If you're using Python 3 you would need:

```
python3 -m pip install rpi.gpio --upgrade
```

But, broadly, this is an issue with `RPi.GPIO` and not something that can be fixed here if the updated version doesn't work.

closed 2020-07-12 by Gadgetoid