pimoroni/fanshim-python #78: [HowTo] Install Fan SHIM on Manjaro ARM

Cyber-Oto, 2020-04-20:
## [HowTo] Install Fan SHIM on Manjaro ARM 

Instructions how to install Fan SHIM, with **_no code_** changes (tested on KDE):

- Install `python-configparser` (5.0.0-1)
- Install latest Fan SHIM
`git clone https://github.com/pimoroni/fanshim-python`  
`cd fanshim-python`  
`sudo ./install.sh`  
- latest version of `psutil` will be installed by `./install-service.sh` script
`cd examples`  
`sudo ./install-service.sh --on-threshold 65 --off-threshold 55 --delay 2`  

```
System:    Kernel: 4.19.114-1-MANJARO-ARM aarch64 bits: 64 compiler: gcc v: 9.2.0 Desktop: KDE Plasma 5.18.4
           tk: Qt 5.14.2 wm: kwin_x11 dm: SDDM Distro: Manjaro ARM
Machine:   Type: ARM Device System: Raspberry Pi 4 Model B Rev 1.1 details: BCM2835 rev: c03111 serial: <filter>

[user@pi4 ~]$ systemctl status pimoroni-fanshim.service
● pimoroni-fanshim.service - Fan Shim Service
     Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: disabled)
     Active: active (running) since Mon 2020-04-20 14:27:04 CEST; 46min ago
   Main PID: 327 (python3)
      Tasks: 2 (limit: 4265)
     Memory: 15.5M
     CGroup: /system.slice/pimoroni-fanshim.service
             └─327 /usr/bin/python3 /home/cyberoto/Projects/fanshim-python/examples/automatic.py --on-threshold 65 --off-threshold>
```

HeadyWook, 2020-05-18:
it should be
`https://github.com/pimoroni/fanshim-python.git`

Cyber-Oto, 2020-05-18:
Both ways produce the same result.

`git clone https://github.com/pimoroni/fanshim-python` results in
```
Cloning into 'fanshim-python'...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 503 (delta 0), reused 2 (delta 0), pack-reused 497
Receiving objects: 100% (503/503), 84.14 KiB | 495.00 KiB/s, done.
Resolving deltas: 100% (307/307), done.
```
`git clone https://github.com/pimoroni/fanshim-python.git` results in
```
Cloning into 'fanshim-python'...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 503 (delta 0), reused 2 (delta 0), pack-reused 497
Receiving objects: 100% (503/503), 84.14 KiB | 638.00 KiB/s, done.
Resolving deltas: 100% (307/307), done.
```
I used the listed installation steps from `README.md` to create this short howto.

Crazyk031, 2020-05-18:
Trying to get this to work on minimal. Haven't been able to iron it out yet.

HeadyWook, 2020-05-19:
> Both ways produce the same result.
>
> `git clone https://github.com/pimoroni/fanshim-python` results in
>
> ```
> Cloning into 'fanshim-python'...
> remote: Enumerating objects: 6, done.
> remote: Counting [...]

I tried this too. I got:
[image]
[image]
[image]

What do you think was going on?

Cyber-Oto, 2020-05-19:
It seems that you are still using Python **2** and missing `pip`. 

You can check what Python package is available, e.g.  `pacman -Qs python-pip`. If there is no output after running the above command, that means that the package is not installed. In order to install it, run `sudo pacman -Syu python-pip`.

**Notes:** 
- all python3 packages are prefixed `python-`, whereas python2 packages are prefixed `python2-`.
- be extra careful _not_ to use `sudo` with pip. _If_ you use `sudo` to install packages using `pip` you may get issues later on when installing packages using `pacman`.

zeldaboch, 2020-05-22:
> Trying to get this to work on minimal. Haven't been able to iron it out yet.
> 
> https://ybin.me/p/7ec74707b57e7ddd#fYmSQw/ItSkvCdyQJXUAWMdKH2RExx/UxkxfcHxSZws=

Same as you, i noticed that the scrip also call for RPi.GPIO that is not present in manjaro repo, but python-raspberry-gpio

theremotheman, 2020-05-25:
> > Trying to get this to work on minimal. Haven't been able to iron it out yet.
> > https://ybin.me/p/7ec74707b57e7ddd#fYmSQw/ItSkvCdyQJXUAWMdKH2RExx/UxkxfcHxSZws=
> 
> Same as you, i also noticed that the scrip also call for RPi.GPIO that is not present in manjaro repo, but python-raspberry-gpio

Run ‘pip install RPi.GPIO’ (without sudo) and You’re good to go

fryfrog, 2020-06-22:
Can't you use the AUR w/ Manjaro? I made a [fanshim](https://aur.archlinux.org/packages/fanshim/) AUR package.

JPUnD, 2020-08-24:
I'm planning on getting a fanshim for myself and I have my Raspberry Pi 4 running Manjaro ARM so I'll check that out myself aswell, although after checking the AUR I have a small question, in your systemd unit file you have `/usr/bin/fanshim` while this github repo as something like `python fanshim`, does that mean that if I were to download from the AUR I would compile the python script named `automatic.py` and use it as a binary? Would that improve performance aswell?

fryfrog, 2020-08-24:
It would not improve performance, it just runs the python script like normal. There is `#!/usr/bin/env python3` at the top, which is why it doesn't *need* to be run like `python <script>`. There is something wrong on `aarch64` though, I *just* put it on one of my Pi 4s and fanshim doesn't think it is on a Pi. Probably something kernel 5.8 related.

JPUnD, 2020-09-07:
```
Sep 07 12:57:17 UD-TinyPiNAS systemd[1]: Started Fan Shim Service.
Sep 07 12:57:17 UD-TinyPiNAS fanshim[513]: Traceback (most recent call last):
Sep 07 12:57:17 UD-TinyPiNAS fanshim[513]:   File "/usr/bin/fanshim", line 2, in <module>
Sep 07 12:57:17 UD-TinyPiNAS fanshim[513]:     from fanshim import FanShim
Sep 07 12:57:17 UD-TinyPiNAS fanshim[513]:   File "/usr/lib/python3.8/site-packages/fanshim/__init__.py", line 1, in <module>
Sep 07 12:57:17 UD-TinyPiNAS fanshim[513]:     import RPi.GPIO as GPIO
Sep 07 12:57:17 UD-TinyPiNAS fanshim[513]: ModuleNotFoundError: No module named 'RPi'
Sep 07 12:57:17 UD-TinyPiNAS systemd[1]: fanshim.service: Main process exited, code=exited, status=1/FAILURE
Sep 07 12:57:17 UD-TinyPiNAS systemd[1]: fanshim.service: Failed with result 'exit-code'.
Sep 07 12:57:17 UD-TinyPiNAS systemd[1]: fanshim.service: Scheduled restart job, restart counter is at 5.
Sep 07 12:57:17 UD-TinyPiNAS systemd[1]: Stopped Fan Shim Service.
Sep 07 12:57:17 UD-TinyPiNAS systemd[1]: fanshim.service: Start request repeated too quickly.
Sep 07 12:57:17 UD-TinyPiNAS systemd[1]: fanshim.service: Failed with result 'exit-code'.
Sep 07 12:57:17 UD-TinyPiNAS systemd[1]: Failed to start Fan Shim Service.
```

I installed from your AUR package, it seems like there's a missing module.

fryfrog, 2020-09-07:
I installed an Arch Linux aarch64 and had some issues, it didn't think it was a Pi. I didn't get much further than that though. :(

JPUnD, 2020-09-10:
I think I've going something, when I run `python` from the terminal as usual I get:
```
Python 3.8.5 (default, Sep  5 2020, 17:18:53) 
[GCC 10.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
```

I thought to myself, if the module isn't found, then maybe it's in the wrong place, otherwise it would return an error.
Python packages, at least to my knowledge, are stored in `/lib/pythonX.X`, after going there I find that I have two versions of Python there - Python 3.7 and Python3.8 - and on the Python 3.7 folder I found this:
```
$ (/lib/python3.7/site-packages) find * | grep RPi
RPi
RPi/GPIO
RPi/GPIO/__pycache__
RPi/GPIO/__pycache__/__init__.cpython-37.pyc
RPi/GPIO/__pycache__/__init__.cpython-37.opt-1.pyc
RPi/GPIO/__init__.py
RPi/__pycache__
RPi/__pycache__/__init__.cpython-37.pyc
RPi/__pycache__/__init__.cpython-37.opt-1.pyc
RPi/__init__.py
RPi/_GPIO.cpython-37m-aarch64-linux-gnu.so
RPi.GPIO-0.7.0-py3.7.egg-info
```

Apparently, they're all being installed as a Python3.7 package, instead of a Python3.8 package. I tried the quick and definitely not recommended approach of just copying the files from `/lib/python3.7` to `/lib/python3.8` but it returned a similar error when running fanshim:
```
$ fanshim --help
Traceback (most recent call last):
  File "/usr/bin/fanshim", line 2, in <module>
    from fanshim import FanShim
  File "/usr/lib/python3.8/site-packages/fanshim/__init__.py", line 1, in <module>
    import RPi.GPIO as GPIO
  File "/usr/lib/python3.8/site-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
ModuleNotFoundError: No module named 'RPi._GPIO'
```

Running Python from the terminal outputs the same:
```
$ python
Python 3.8.5 (default, Sep  5 2020, 17:18:53) 
[GCC 10.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/site-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
ModuleNotFoundError: No module named 'RPi._GPIO'
```

fryfrog, 2020-09-10:
You should be able to get them in the right place by re-installing the dependencies.

JPUnD, 2020-09-11:
Okay I've figured it out, apparently if you're running Manjaro ARM there's a package in the official repositories that's named `python-raspberry-gpio` which is still in Python 3.7. The workaround is to install that package straight from the AUR using the `--aur` flag: `yay --aur python-raspberry-gpio`

However now I'm running into compiling issues. I'll also post this in the Manjaro ARM forums because they have an outdated package anyways.

If you want to help with the debugging of the compiling errors here's the error:
```
collect2: error: ld returned 1 exit status
error: command 'gcc' failed with exit status 1
==> ERROR: A failure occurred in package().
    Aborting...
error making: python-raspberry-gpio
```

And the full log here over at [pastebin](https://pastebin.com/JdefjAm3)

JPUnD, 2020-09-11:
After a discussion over at the Manjaro ARM forums [here](https://forum.manjaro.org/t/outdated-package-python-raspberry-gpio/24045) I've found a fix for the issue, simply changing the `PKGBUILD` of the package `python-raspberry-gpio` in the following manner.
Run:
```
yay --aur --editmenu -S python-raspberry-gpio
```
And then change the file:
```
python setup.py install --prefix=/usr --root="$pkgdir" --optimize=1
```
To:
```
env CFLAGS="-fcommon" python setup.py install --prefix=/usr --root="$pkgdir" --optimize=1
```

The Manjaro ARM team said he will rebuild the package and update the repos today :).

After commenting on the [python-raspberry-gpio](https://aur.archlinux.org/packages/python-raspberry-gpio/) AUR page, the maintainer said the original developer as lost interest on the package and it's using `sysfs core interface` which is deprecated software and will soon be removed, so he recommended to use `libgpiod` instead.

All and all, I've managed to finally install and have `fanshim` up and running on my Raspberry Pi 4 and hopefully everyone can have too.

JPUnD, 2020-09-12:
Alright running `yay -S fanshim` and then doing `sudo systemctl enable --now fanshim` enables the fanshim service on a Raspberry Pi 4 if someone wants to use the AUR to use this it's working for me!

I only have a small question to @fryfrog, I wanted to make a small customization to the script to have the Pi shutdown if it reaches a high enough temperature (if the fan were to fail or something) what's the recommended way of doing so to avoid an update from the AUR ruining my modifications?

fryfrog, 2020-09-12:
Is `libgpiod` a drop in replacement for `python-raspberry-gpio`?

For the script customization, why don't you write it as a `--option` and then pull request it to fanshim?

JPUnD, 2020-09-12:
> Is `libgpiod` a drop in replacement for `python-raspberry-gpio`?
> 
> For the script customization, why don't you write it as a `--option` and then pull request it to fanshim?

No from what I understood `libgpoid` is a replacement for the library `RPi.GPIO` which according to the AUR maintainer of `python-raspberry-gpio` is unmaintained, I check the [homepage](https://pypi.org/project/RPi.GPIO/#history) in PyPI and the last update was more than a year ago.
Also there's already a project which is trying to have `fanshim` in C++ working using `libgpoid` over [here](https://github.com/daviehh/fanshim-cpp) on Github (it's also on the bottom of the readme on this github page) but it hasn't seen an update in a year and it's currently experimental.

As for the suggestion, yeah I think that's a good idea, I know some Python this would be a good oportunity for me to actually contribute something other than just reporting and trying to fix bugs. Thanks!

JPUnD, 2020-09-12:
I think you can close this issue and add the install instructions for Manjaro/Arch ARM, no?

likenedthus, 2020-11-09:
Would this be similar to how I'd go about getting fanshim to work on Ubuntu 20.10 64-bit?

carlosrberto, 2020-11-23:
@likenedthus I had problems to make it work on Ubuntu 20.10 64 bits, `pip install rpi.gpio` was not working for me. Solved by running:
```
sudo apt-get install python3-rpi.gpio
```

jpmvferreira, 2021-05-21:
Okay here's an update quite a bit latter.

I left my RPI turned on for a few months and only shut it down a couple of times in the last weeks, and then this week I noticed that the fan was always on, and I'm not sure when it decided to turn on by itself.

I updated my RPI which I haven't in a while and then fanshim gave an error, so considering the conflict of the package `python-raspberry-gpio` between Manjaro ARM repository and AUR I decided to uninstall the entire thing and reinstall it.

So here's what I did:
```bash
$ yay -Rns fanshim 
(...)
$ yay --aur -S python-raspberry-gpio
(...)
$ yay -S fanshim
```

Now when running `fanshim` I get:
```bash
$ fanshim
Traceback (most recent call last):
  File "/usr/bin/fanshim", line 92, in <module>
    fanshim = FanShim(disable_button=args.nobutton, disable_led=args.noled)
  File "/usr/lib/python3.9/site-packages/fanshim/__init__.py", line 32, in __init__
    GPIO.setup(self._pin_fancontrol, GPIO.OUT)
RuntimeError: Not running on a RPi!
```

The fan is stopped and the LED is solid green but I don't think I can control anything, any ideas on how to proceed?

jpmvferreira, 2021-05-22:
```bash
$ fanshim
Traceback (most recent call last):
  File "/usr/bin/fanshim", line 92, in <module>
    fanshim = FanShim(disable_button=args.nobutton, disable_led=args.noled)
  File "/usr/lib/python3.9/site-packages/fanshim/__init__.py", line 32, in __init__
    GPIO.setup(self._pin_fancontrol, GPIO.OUT)
RuntimeError: Not running on a RPi!
```

Apparently this is a permission related issue, using root privileges I can run this as expected, which means that the previous installation worked.

jpmvferreira, 2023-07-30:
I would like to point out that the fanshim package in the AUR got removed.

If somebody wants to ports this package to the AUR, making this installation process easy on Arch based distros, they would have to package this program as an AUR package and the apa102 Python library.
The RPi.GPIO package is already available in the AUR and is working on my RPI4B running Manjaro ARM.
I don't think there any extra dependencies to take care of.