pimoroni/fanshim-python #86: shim not compatible with heatsinks

sk8nfool, 2020-06-25:
I recently purchased an RPi-4B that came with a set of heatsinks (or the simple "case" did - don't remember). The fan shim isn't compatible with the heat sinks. It (the fan shim) hits the CPU heat sink (and the SDRAM heat sink) preventing it from being properly seated on the GPIO pins. This fact should be clearly stated on all fan shim documentation.

druck13, 2020-06-27:
On 25/06/2020 22:01, sk8nfool wrote:
> I recently purchased an RPi-4B that came with a set of heatsinks (or the
> simple "case" did - don't remember). The fan shim isn't compatible with
> the heat sinks. It (the fan shim) hits the CPU heat [...]

It seems to say this pretty clearly at the bottom of

https://shop.pimoroni.com/products/fan-shim

"﻿Not heatsink-compatible!"

Cheers
---David
-- 
Email: druck@druck.org.uk
Phone: +44-(0)7974 108301

Gadgetoid, 2020-07-12:
While I appreciate the issue, this is clearly not a software problem and thus is outside the scope of GitHub issues.

I will say- however- that Fan SHIM can easily have a pin-header attached to raise it above a heatsink. Obviously a bit more tricky that attaching it straight out of the box, but it should work.

closed 2020-07-12 by Gadgetoid

Petros626, 2023-02-15:
@sk8nfool it is, but you just have to make some changes. In my case I ordered the standard heatsink für Raspberry Pi 4B and place them on the board. You only need to cut one heatsink (for RAM) in two pieces and stick one piece on the half of the RAM.