pimoroni/fanshim-python #87: LEDs are not working

Michael-Axelsson, 2020-07-12:
Hi, sorry, very new in this game but thought that the Fan SHIM would be better compared to a standard fan that runs permanently
I installed the Fan SHIM and then followed the instructions for installing the software (https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-fan-shim). it seems to run fine but the LED do not show any color (red o green) is this a problem or something that I just missed in the instructions?

Gadgetoid, 2020-07-12:
My first guess would be poor electrical contact between Fan SHIM and your Pi's GPIO pins. Try giving Fan SHIM enough of a wiggle to make contact and see if it lights up.

It might also be worth running the LED example in examples to see if it works: https://github.com/pimoroni/fanshim-python/blob/master/examples/led.py

Michael-Axelsson, 2020-07-12:
Hi Philip

Thank you for rapid feedback, can I do this with the RPI on?

Sincerely
Michael

Michael-Axelsson, 2020-07-12:
Hi again

I tried to rest/wiggle the Fan SHIM but no LED turns on, the fan seems to be doing what it should cycling of and on depending on the RPI temperature (sudo ./install-service.sh --on-threshold 40 --off-threshold 35 --delay 2) I did select 40 and 35 just to speed up testing

I ran led.py (sudo python3 led.py), it stops the fan but nothing else
I ran the toggle.py (it also stops the fan) and that works, I can start and stop he fan with the push button

Sincerely
Michael

Gadgetoid, 2020-07-13:
Cheers for trying. This sounds like a hardware issue, you should drop an email to support@pimoroni.com since they'll be able to replace if they don't figure out a fix.

closed 2020-07-13 by Gadgetoid

Michael-Axelsson, 2020-07-13:
Hi

Thank you for you support, I will contact pimoroni even  if I have no ise for the LED in the final setup since it will be encased and the fan not visible

I have a number of other RPI running and will buy more of these since I like the idea of a fan that only runs when needed

Cheers
Michael