pimoroni/fanshim-python #89: Preempt doesnt work with 32bit userland 64bit kernel

puccaso, 2020-07-27:
Hey guys, 

the preempt feature doesn't seem to work with 32bit userland 64bit kernel on raspberry pi os / rpi 4b.
when i use with the preempt, everything is ok but even though th preempt option doesn't give me errors, when i go over the temperature threshold, it the fan does not kick in.

and i'm running on a headless system - so if i hadn't randomly VNCd into the machine i wouldn't have seen the temp warning!

any idea how to fix?

puc

Gadgetoid, 2020-07-27:
This would imply that `psutil.cpu_freq()` does not return the correct current frequency.

You could probably verify that by running the following both during idle, and during some load (like `sysbench --test=cpu --cpu-max-prime=20000 run`):

```python
import psutil
import time

time.sleep(1.0)
print(psutil.cpu_freq().current)
```

On my Pi 4 with stock Raspberry Pi OS, kernel 5.4.51 this will show `600` at idle and `1500` at load, it's this transition that pre-empt uses to try and kick in cooling before the CPU actually starts to heat up.

In all cases, though, as soon as the system *does* heat up and exceed the temperature threshold, the cooling should kick in so something else is amiss here.

scollar71, 2020-11-30:
Just purchased an Argon 1 M.2 (Yet to recieve) I believe it uses a identical setup as Pimoroni(?) has there been any work to adapt to 64bit OS's. Currently loving the lineage OS 17.1 Android 10 TV setup on my pi 4b am I going to have headaches with that OS?

closed 2023-01-20 by Gadgetoid