pimoroni/fanshim-python #94: LED control doesn't appear to work

jimklo, 2021-01-06:
Trying to create my own variant of the service that controls the led in a different manner.

Finding that the library doesn't seem to alter how the LED functions.

```
import fanshim import FanShim

fs = FanShim()
fs.set_light(0, 0, 0, brightness=0)  # assuming this should turn it off, however led stays lit.
fs.set_light(255, 0, 0, brightness=0.5) # assuming this should turn the led red, at about 50%  brightness, however led stays lit white.
```

jimklo, 2021-01-06:
Okay, I think I may know what's going on.  Looking at the [shim pinout](https://pinout.xyz/pinout/fan_shim) and comparing with the 
[image], I'm seeing that there is a conflict with GPIO 14/15.  Not sure if there is a way to modify the shim to make this work short of having another shim board and wiring to a pair of unused GPIO pins.

Gadgetoid, 2022-02-14:
Some trace cutting and careful bodge wire soldering might work, but otherwise you'd have to spin the fan shim around, wire up to the header manually and find some way to secure it to the board.

I think in the time since we originally created Fan SHIM the Pi 4's thermal management has been "fixed" anyway, so it's probably redundant.

closed 2022-02-14 by Gadgetoid