pimoroni/fanshim-python #95: Fan Shim causing boot errors

ToastedVegetables, 2021-03-15:
Hello,
The Fan Shim is causing boot errors on a Raspberry Pi 4. 
Specifically, when using Ubuntu Server 20.04.2.0 LTS and 20.10, the normal boot sequence is interrupted and I get various errors such as "No Network USB Devices Found", after which the Raspberry Pi seems to spam and delete random characters into the terminal. In addition to this, the boot screen/logo flashes across the screen, making me unable to login.
During all of this, the Fan Shim ramps up and slows down seemingly at random. I think that this may be caused by some interference between the GPIO pins and the Ubuntu server software, or between the GPIO and the keyboard that I had connected to the Pi. 
Removing the Fan Shim solves the problem, but I'm curious as to why the fan shim is interfering with server boot procedures- I have heard of some OS errors caused by GPIO devices attached to the Pi upon boot.

Gadgetoid, 2021-03-15:
This sounds suspiciously like UART is enabled on GPIO14 and GPIO14. These are used for LED control on Fan SHIM and could easily be read as UART input with more or less the effects you describe.

ToastedVegetables, 2021-05-19:
I've solved this issue by simply not using the fan shim, but that's not really a "solution" per se. How would I disable UART on my Raspberry Pi GPIO?

Gadgetoid, 2022-02-14:
(Sorry I'm late to reply here) You should be able to disable it in (IIRC) "Interfacing Options" in raspi-config or the Raspberry Pi Configuration GUI.

closed 2022-02-14 by Gadgetoid