pimoroni/fanshim-python #96: automatic.py doesn't detect (installed) RPi module

dchmelik, 2021-03-23:
On Slackware ARM for Raspberry Pi (SARPi) current version, automatic.py doesn't detect (installed) RPi module:
```
root@1.darwinheim:~/fanshim-python/examples# pip3 install RPi
Requirement already satisfied: RPi in /usr/lib/python3.9/site-packages (0.0.1)
root@1.darwinheim:~/fanshim-python/examples# ./automatic.py
Traceback (most recent call last):
  File "/root/fanshim-python/examples/./automatic.py", line 2, in <module>
    from fanshim import FanShim
  File "/usr/lib/python3.9/site-packages/fanshim/__init__.py", line 1, in <module>
    import RPi.GPIO as GPI
ModuleNotFoundError: No module named 'RPi'
```
Perhaps it depends on something else?  Older programmers have called Python a mess of dependencies; is there a dependency list?
On another topic, I use my RPi with a HighPi case (has room for a hat) and serial port hat but want to put the fan underneath it but couldn't find anything called 'booster header'... what should I use for that, exactly, or maybe the serial port hat doesn't need to be all the way on the pins?  If it does, I'm unsure both will fit...

Gadgetoid, 2022-02-14:
Sorry I'm probably far too late to be of any use here, but I suspect "pip3" is probably referring to a version of python *other than* the "python3" that Fan SHIM uses. This could be because you have multiple versions of Python 3 installed, or still running in a virtual environment, or some other mystery meat quirk of Python.

I tend to be explicit with: `python3 -m pip install rpi.gpio` to avoid this, but... yeah Python is still a nightmare :grimacing:

closed 2022-02-14 by Gadgetoid

dchmelik, 2022-02-15:
There neither were multiple Python 3 (though has Python 2) nor virtualization.  However your longer command worked.