pimoroni/fanshim-python #97: Fan does not go off if the PI is powered down

neilyoung, 2021-04-01:
As subject says. I'm having the service running to control the fan, but if I `sudo shutdown now` the PI powers off, but the fan still roars. Only a complete removal of power from the PI switches off the fan.  Is that by design?

jpmvferreira, 2021-05-17:
I'm getting the same problem, it's probably by design, and I'm unsure if it's a hardware thing or not because it's connected directly to the pins.

But I would also like to see it powering down and the RPI is powered off!

Gadgetoid, 2022-02-14:
This is by design, but you may be able to force it off by asserting a GPIO pin on shutdown.

It'll require a script in `/lib/systemd/system-shutdown/gpio-poweroff` that looks something like this - https://github.com/pimoroni/clean-shutdown/blob/master/daemon/lib/systemd/system-shutdown/gpio-poweroff

And drop a config file in `/etc/cleanshutd.conf` with:

```
poweroff_pin=18
```

cmonty14, 2022-03-04:
> This is by design, but you may be able to force it off by asserting a GPIO pin on shutdown.
>
> It'll require a script in `/lib/systemd/system-shutdown/gpio-poweroff` that looks something like this - [...]

I cannot confirm that this solution is working, means 120s after shutdown the fan is still running.
Maybe I missed a step.
Can you please advise what you mean by "[...] asserting a GPIO pin on shutdown."?

yrjoseppa, 2025-02-10:
The most simple way to overcome the problem is to put an external 10kohm pull-down resistor between GPIO 18 and GND. To be sure that it works (since the fan is most if the time off now) I made a stress test for my Raspberry Pi 4 Model B and got the processor heated up. After reaching the pre-defined on-threshold temperature the fan started and on pre-defined off-threshold shut down as expected. And most importantly the fan stays now off after the shutdown even the Raspberry board get it's voltages from the power adapter (and keeps the red LED on). This was the only solution that I have found so far :-)

[image]