pimoroni/fanshim-python #100: install.sh fails on ubuntu server 21.04

clancia, 2021-06-04:
Hi there

I tried to install fanshim on my RPi4 running ubuntu server 21.04 (64-bit)
This OS is not shipped with python 2.x
I had troubles with the install script, which would not manage to install rpi.gpio
So I installed it manually via `pip3 install RPi.GPIO`
I could import the package in the interpreter but still the `install.sh` script  would try to install rpi.gpio anyway and the installation would not be successful

Then I let apt install rpi.gpio via `sudo apt install python3-rpi.gpio` and voilá, it works.

clancia, 2021-06-04:
```
Checking for rpi.gpio >= 0.7.0 (for Pi 4 support)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'RPi'
Installing rpi.gpio
Collecting rpi.gpio>=0.7.0
  Using cached RPi.GPIO-0.7.0.tar.gz (30 kB)
Building wheels for collected packages: rpi.gpio
  Building wheel for rpi.gpio (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-nwmca41p/rpi-gpio_da816315d76d409a90ebcd976717f7f5/setup.py'"'"'; __file__='"'"'/tmp/pip-install-nwmca41p/rpi-gpio_da816315d76d409a90ebcd976717f7f5/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-xcil89aa
       cwd: /tmp/pip-install-nwmca41p/rpi-gpio_da816315d76d409a90ebcd976717f7f5/
  Complete output (102 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-3.9
  creating build/lib.linux-aarch64-3.9/RPi
  copying RPi/__init__.py -> build/lib.linux-aarch64-3.9/RPi
  creating build/lib.linux-aarch64-3.9/RPi/GPIO
  copying RPi/GPIO/__init__.py -> build/lib.linux-aarch64-3.9/RPi/GPIO
  running build_ext
  building 'RPi._GPIO' extension
  creating build/temp.linux-aarch64-3.9
  creating build/temp.linux-aarch64-3.9/source
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/c_gpio.c -o build/temp.linux-aarch64-3.9/source/c_gpio.o
  source/c_gpio.c: In function ‘setup’:
  source/c_gpio.c:130:9: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
    130 |     if ((uint32_t)gpio_mem % PAGE_SIZE)
        |         ^
  source/c_gpio.c:131:34: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
    131 |         gpio_mem += PAGE_SIZE - ((uint32_t)gpio_mem % PAGE_SIZE);
        |                                  ^
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/common.c -o build/temp.linux-aarch64-3.9/source/common.o
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/constants.c -o build/temp.linux-aarch64-3.9/source/constants.o
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/cpuinfo.c -o build/temp.linux-aarch64-3.9/source/cpuinfo.o
  source/cpuinfo.c: In function ‘get_rpi_info’:
  source/cpuinfo.c:139:28: warning: format ‘%llx’ expects argument of type ‘long long unsigned int *’, but argument 3 has type ‘uint64_t *’ {aka ‘long unsigned int *’} [-Wformat=]
    139 |       sscanf(revision, "%llx", &rev);
        |                         ~~~^   ~~~~
        |                            |   |
        |                            |   uint64_t * {aka long unsigned int *}
        |                            long long unsigned int *
        |                         %lx
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/event_gpio.c -o build/temp.linux-aarch64-3.9/source/event_gpio.o
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-xSUlVY/python3.9-3.9.5=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c source/py_gpio.c -o build/temp.linux-aarch64-3.9/source/py_gpio.o
  source/py_gpio.c: In function ‘PyInit__GPIO’:
[... cut ...]
  /usr/bin/ld: build/temp.linux-aarch64-3.9/source/soft_pwm.o:/tmp/pip-install-nwmca41p/rpi-gpio_da816315d76d409a90ebcd976717f7f5/source/soft_pwm.c:28: multiple definition of `threads'; build/temp.linux-aarch64-3.9/source/event_gpio.o:/tmp/pip-install-nwmca41p/rpi-gpio_da816315d76d409a90ebcd976717f7f5/source/event_gpio.c:60: first defined here
  collect2: error: ld returned 1 exit status
  error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for rpi.gpio
  Running setup.py clean for rpi.gpio
Failed to build rpi.gpio

closed 2021-06-04 by clancia

RobinTTY, 2021-06-06:
Just encountered the same problem, thanks for sharing the solution!