pimoroni/fanshim-python #101: Fan software not working

Zelatrix, 2021-07-14:
I have a Pimoroni fan shim, and when it is plugged in, it works, and cools my Pi. But if I install the fanshim service, while I am able to control the LED using the `led.py` and `led2.py` appears to work, if I write my own scripts to control the LED or fan, or try to control it manually by running `manual.py` and issuing a command like `echo "FF0000" > /tmp/fanshim` doesn't affect the fan or the LED.

Gadgetoid, 2022-02-14:
You're probably getting fighting between the service trying to control the LED and your scripts. `led.py` and `led2.py` only win because they update the LEDs at 60FPS - https://github.com/pimoroni/fanshim-python/blob/33abcde2d51944e6dcb3bc2ecbddb251f65eb79a/examples/led.py#L10-L14

Install the service with the `--noled` option.

Looks like the README need updating to mention this.

Gadgetoid, 2022-02-14:
Closing in favour of #99 to track the documentation issues

closed 2022-02-14 by Gadgetoid