pimoroni/fanshim-python #106: commands work but fan doesn't start

dchmelik, 2022-02-15:
On Slackware-ARM32 15, installation (after uncommenting Debian CODENAME lines from current commit install.sh) and commands work, but don't actually start fan.  Here's output.  Should I get install log also?  Maybe it only works on RaspiOS?
```
root@1.darwinheim:~# python3
Python 3.9.10 (main, Jan 17 2022, 21:15:54) 
[GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from fanshim import FanShim
>>> fanshim = FanShim()
>>> fanshim.set_fan(True)
True
```
That was on a Raspberry Pi (RPi) 4.  On a RPi3 (but same OS) there's a different problem.

closed 2023-01-20 by Gadgetoid

dchmelik, 2023-01-21:
This wasn't completed!

Gadgetoid, 2023-01-24:
If it's not Raspberry Pi OS I'm afraid I can't be of much help. If we support one alternate OS we'd have to support them all. Sorry.

The fan should default to on, IIRC, though, so you might have a service or driver that's messing with BCM 18 and preventing control. This library is just toggling a GPIO pin.