pimoroni/fanshim-python #108: Fan always on and not responding to button or automatic code

Kevin-Mattheus-Moerman, 2022-03-01:
1. Similar to https://github.com/pimoroni/fanshim-python/issues/97 the fan is always on, even after a shutdown. 
2. The led is mostly not on. I do have the fan on the right pins but perhaps the contact is not good? I have tried to reapply the fan e.g. slightly slanted or pressed further down. Sometimes the LED is bright white, which seems wrong, or sometimes green but it isn't very consistent and it is mostly off. 
3. If I press the button long (or then short as instructed) nothing happens. When the green light is on (only sometimes see point 2) I think the button responds, or at least I think I've seen the LED switch off due to the button, but the fan stays on even if I follow the button sequence to turn off the fan. 

Any help or guidance to trouble shoot this would be appreciated. Thanks.

Kevin-Mattheus-Moerman, 2022-03-19:
@Gadgetoid :wave: any thoughts?

Bursucel, 2022-11-30:
Make sure that the shim makes good contact with all the pins. I had a similar issue and turned out to be a bad contact.
In my case the default behavior would be that the fan would turn on at boot up and would not turn off when the service started,just kept going on.
I've re seated the shim. I have it sandwiched between the pi gpio pins and an extension cable. Once I got a good contact everything worked.
I hope it helps.