pimoroni/fanshim-python #109: doesn't install/detect on some systems

dchmelik, 2022-03-02:
On SlackwareARM 15 on Raspberry Pi (RPi) 3 fanshim-python doesn't detect what's installed (configparser, six) and doesn't finish installing.  On SlackwareARM 15 on RPi4 (identical installation) it installs fine (but doesn't run the fan, other ticket.)
I made an [installation error log](https://gist.github.com/dchmelik/799aaad96384a6fd11c311643e25d28f) where I showed configparser, six are installed but at the end there's no /usr/lib/python3.9/site-packages/fanshim.

dchmelik, 2023-01-21:
This wasn't completed!

Gadgetoid, 2023-01-24:
This is very intentional. We only support Raspberry Pi OS. There are simply too many alternate operating systems out there to handle the myriad edge cases.

You can, in most cases, just `python3 -m pip install fanshim` to *just* get the Python library though.