pimoroni/fanshim-python #112: Service installation hangs on Raspberry Pi OS Lite 64bit April 4th 2022

GarikFirst, 2022-05-05:
I am on fresh new Raspberry Pi OS Lite 64bit April 4th 2022:

```
git clone https://github.com/pimoroni/fanshim-python.git
cd fanshim-python/
sudo ./install.sh 
cd examples/
sudo ./install-service.sh 

```
Installation just hangs:
```
pi@raspberrypi:~/fanshim-python/examples $ sudo ./install-service.sh 
Setting up with:
Off Threshold:    55 C
On Threshold:     65 C
Low Temp:         55 C
High Temp:        65 C
Delay:            2 seconds
Preempt:          no
Disable LED:      no
Disable Button:   no
Brightness:       255
Extended Colours: no

To change these options, run:
sudo ./install-service.sh --off-threshold <n> --on-threshold <n> --delay <n> --brightness <n> --low-temp <n> --high-temp <n> --venv <python_virtual_environment> (--preempt) (--noled) (--nobutton) (--extended-colours)

Or edit: /etc/systemd/system/pimoroni-fanshim.service

Checking for rpi.gpio >= 0.7.0 (for Pi 4 support)
rpi.gpio >= 0.7.0 already installed
Checking for Fan SHIM
Fan SHIM already installed
Checking for psutil >= 5.6.7
psutil >= 5.6.7 already installed

Installing service to: /etc/systemd/system/pimoroni-fanshim.service
```

If I do it by hands, eg: create unit file for service, then trying to start this service by systemctl - it just hangs on service starts.

Running automatic.py manually works fine:
```
pi@raspberrypi:~/fanshim-python/examples $ python3 automatic.py --verbose
Current: 39.43 Target: 55.00 Freq  1.50 Automatic: True On: False
Current: 40.41 Target: 55.00 Freq  1.50 Automatic: True On: False
```

GarikFirst, 2022-05-11:
It somehow started working as it should

closed 2022-05-11 by GarikFirst

duc00, 2024-01-14:
I have this issue as well