pimoroni/fanshim-python #114: The fan does not work

homerybart, 2022-07-31:
**Hi**
 
  **When I check the status of the script with systemctl it shows this:**

  _systemctl status pimoroni-fanshim.service
● pimoroni-fanshim.service - Fan Shim Service
   Loaded: loaded (/etc/systemd/system/pimoroni-fanshim.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2022-07-31 10:58:22 CEST; 3s ago
  Process: 11680 ExecStart=/usr/bin/python3 /home/pi/automatic.py --on-threshold 65 --off-threshold 55 --delay 2 --brightness 255 (code=exited, status=2)
 Main PID: 11680 (code=exited, status=2)

jul 31 10:58:22 raspberryPi systemd[1]: pimoroni-fanshim.service: Service RestartSec=100ms expired, scheduling restart.
jul 31 10:58:22 raspberryPi systemd[1]: pimoroni-fanshim.service: Scheduled restart job, restart counter is at 5.
jul 31 10:58:22 raspberryPi systemd[1]: Stopped Fan Shim Service.
jul 31 10:58:22 raspberryPi systemd[1]: pimoroni-fanshim.service: Start request repeated too quickly.
jul 31 10:58:22 raspberryPi systemd[1]: pimoroni-fanshim.service: Failed with result 'exit-code'.
jul 31 10:58:22 raspberryPi systemd[1]: Failed to start Fan Shim Service._

**I have this version of Python installed:** 

_2.7.16 (default, Oct 10 2019, 22:02:15)
[GCC 8.3.0]_

homerybart, 2022-07-31:
before it worked fine
any help will be appreciated

Regards

homerybart, 2022-08-01:
and the versión of Python3 is: 

3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0]

homerybart, 2022-09-15:
Pimoroni sells its products but then does not give a support, I will not buy anything else from them

helgibbons, 2022-09-15:
Sorry about the slow response! Emailing support@pimoroni.com is usually the fastest way to contact us with support requests  - our software team is small and very busy.

What OS are you using? Is the fan always on, or always off?

Does reinstalling the service help? https://learn.pimoroni.com/article/getting-started-with-fan-shim
You could also try a fresh OS install in case something's got out of sync?

homerybart, 2022-09-15:
Hi!

This is mi OS

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

the fan is always off from two years ago